zigpy/zha-device-handlers #2868: [Device Support Request]

itwtds, 2023-12-28:
### Problem description

New Namron Zigbee Touch Termostat 2.0 is identified as TS0601 by _TZE204_p3lqqy2r with no controls or diagnostic.

https://www.namron.com/products/namron-zigbee-touch-termostat-16a-sort-925 / 4512753
https://www.namron.com/products/namron-zigbee-touch-termostat-16a-hvit-155 / 4512752

### Solution description

The device should have similar support as it's predecessor https://www.namron.com/products/namron-zigbee-touch-termostat-16a-sort / 4512738.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_p3lqqy2r",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.12.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Oslo",
    "os_name": "Linux",
    "os_version": "6.1.63-haos",
    "supervisor": "2023.12.0",
    "host_os": "Home Assistant OS 11.2",
[... 347 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

Jellyfrog, 2024-01-14:
Support was added in zigbee2mqtt, I might give it a shot to implement this during the week. 
https://github.com/Koenkk/zigbee-herdsman-converters/pull/6811

miahn, 2024-01-28:
@Jellyfrog do you need any help testing it?

Jellyfrog, 2024-01-28:
Sadly I decided to give z2m a shot after some other unrelated troubles, so I won't be working on this

vegardoveras, 2024-02-26:
I've tried to build my first quirk for this termostat, but can't get my head around it. I think the guide for making a quirk is a bit too general. Do we have a example for a tuya quirk and how to figure out the input/output?

Kjelllis, 2024-05-06:
> @miahn @Jellyfrog do you need any help testing it?

Hi. I would love some help to get this supported in ZHA for everyone or with a custom quirk.
I have installed this thermostat but can't control it. I can see that Z2M has support now.

eeriksen, 2024-07-12:
In need support for this one too. @itwtds How you found a way to control this thermostat with ZHA?

hansarh, 2024-08-13:
Would be awesome to get this supported! the product URL's are not working btw. I believe this is the new one: 
https://www.namron.com/products/namron-zigbee-touch-termostat-16a-hvit-23/

itwtds, 2024-08-13:
> In need support for this one too. @itwtds How you found a way to control this thermostat with ZHA?

Hei, Eivind.

Sorry about the late reply, I went with z2m as the support was already there.

hansarh, 2024-08-15:
I've tried looking into if there's semi-simple way of creating a quirk by looking at what has been added to the zigbee2mqtt "parsing" [here](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6811/files#diff-db8cc5f8f5a29a734d39a1bbd06ac727adfe050b2f51195c14a84f92f68b3583) and map it to how a quirk is built in this repository. Can anyone provide some guidelines for this? or is this not the way to do it?

mellowism, 2024-11-02:
I would very much like to get this added as well, or help with a quirk. Just bought a couple thinking it would work the same as it's predecessor but sadly no.

Beastware-666, 2024-11-11:
I have the same issue with this one.  (TS0601 by _TZE204_p3lqqy2r) Not showing any controls in ZHA.

closed 2024-12-22 by TheJulianJES