zigpy/zha-device-handlers #2878: [Device Support Request] _TZE200_n9clpsht TS0601 keypad

besiktas97, 2024-01-02:
### Problem description

I can add the device with succes but can't see entities or events.
Its this keypad: https://zigbee.blakadder.com/Immax_07505L.html

### Solution description

I have found some custom quirks but those arent working.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.12.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Amsterdam",
    "os_name": "Linux",
    "os_version": "6.1.21-v8",
    "supervisor": "2023.12.0",
    "host_os": "Home Assistant OS 10.4",
    "docker_version": "23.0.6",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "meross_lan": {
      "version": "4.5.1",
      "requirements": []
    },
    "plant": {
      "version": "2023.11.0",
      "requirements": [
        "async-timeout>=4.0.2"
      ]
    },
    "alarmo": {
      "version": "v1.9.13",
      "requirements": []
    },
    "midea_ac_lan": {
      "version": "v0.3.22",
      "requirements": []
    },
[... 373 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

Albandu51, 2024-01-20:
I am also intereste. I have to stay in HZA. Can anybody help us ??? Thx

besiktas97, 2024-01-20:
Its not relevant for me anymore, because I am using z2m now and everything is working fine now. In this case issue is closed for me, but can stay open for someone else.