zigpy/zha-device-handlers #2886: [Device Support Request] Aqara T1 LED Strip

jaredhamilton, 2024-01-05:
### Problem description

Light strip is available to add via ZHA directly, however you cannot control the length of the strip. When using extensions on the light strip, standard color temperatures will illuminate the entire strip, however using RGB colors will only illuminate the original strip length, minus the extensions.

It should be noted that there are reports of this strip working with Z2M, though the attributes need to be configured through MQTT. See this thread: https://community.home-assistant.io/t/unable-to-turn-on-extensions-for-aqara-t1-led-strip-zha/650487/4

### Solution description

Develop a quirk that includes the strip length attribute.

lindhe, 2024-07-05:
I'd love to see this not be stale. I have a T1 LED strip, is there anything I can do to help?

benjaml4, 2024-07-26:
I to have a T1 LED strip. Unfortunately I don't have any knowledge of programming quirks. Can someone please take a look at the Zigbee2MQTT handler and see if we can map this to ZHA? It would make it so much better removing my Aqara E1 hub as it's causing interference with my default ZHA hub.

skypyxis, 2024-10-12:
It should be possible to set the LED strip color in automations.
While the control panel allows color selection, it's unclear how to achieve this within an automation. Is this possible? How?
Additionally, scenes and effects don't seem to be supported with ZHA

skypyxis, 2024-10-14:
Additionally: what is the purpose of the 'Transition' setting?
It doesn't appear to have any effect. Shouldn't it be used to fade one color into another?

[image]

Ros-One, 2024-12-05:
Same issue.

Is there anything that can be done to set the device length other than using an Aqara hub or using z2m? I installed the zha-toolkit, but didn't manage to figure it out. I expected to be able to send some write_command and set the length to 3m.

gafderks, 2025-02-04:
@Ros-One 
I managed to get this working.

1. Install [ZHA Toolkit](https://my.home-assistant.io/redirect/hacs_repository/?owner=mdeweerd&repository=zha-toolkit&category=integration) in HACS
2. In the Developer Tools run action "ZHA 🧰 Toolkit: Write Attribute" with the following YAML config:

```yaml
action: zha_toolkit.attr_write
data:
  ieee: [entity]
  attribute: 0x51b
  cluster: 0xfcc0
  manf: 4447
  endpoint: 1
  attr_val: 15
```

Here, `attr_val` is the number of segments that the led strip has, which is 10 for the base strip and 5 additional per extension strip.

gmeye, 2025-03-06:
> [@Ros-One](https://github.com/Ros-One) I managed to get this working.
> 
> 1. Install [ZHA Toolkit](https://my.home-assistant.io/redirect/hacs_repository/?owner=mdeweerd&repository=zha-toolkit&category=integration) in HACS
> 2. In the Developer Tools run action "ZHA 🧰 Toolkit: Write Attribute" with the following YAML config:
> 
> action: zha_toolkit.attr_write
> data:
>   ieee: [entity]
>   attribute: 0x51b
>   cluster: 0xfcc0
>   manf: 4447
>   endpoint: 1
>   attr_val: 15
> Here, `attr_val` is the number of segments that the led strip has, which is 10 for the base strip and 5 additional per extension strip.
> 
> Edited to note that I found the cluster and the attribute in the [zigbee-herdsman-converters repository](https://github.com/Koenkk/zigbee-herdsman-converters/blob/b91e0a8e60be1850f9d6bf10cbf47c23f10e0dcc/src/devices/lumi.ts#L326). There, also the attributes for audio, audio sensitivity, as well as some others are listed, I did not test those.

Can you explain how you got that action to run?

I have ZHA toolkit installed, when I go into developer tools and try to run that action I get the following error: Action zha_toolkit.attr_write not found.

Menelmacar, 2025-09-08:
Contribute