zigpy/zha-device-handlers #2889: ALAB CO2 sensor

frootblaster, 2024-01-07:
### Problem description

Alab CO2 sensor displays incorrect CO2 values @9zigen

### Solution description

Display accurate CO2 values

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4660, maximum_buffer_size=108, maximum_incoming_transfer_size=1617, server_mask=11264, maximum_outgoing_transfer_size=1617, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0302",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0402",
        "0x0405",
        "0x040d"
      ],
      "output_clusters": []
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "Alab",
  "model": "Alab-CO2-1.1",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.1.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Stockholm",
[... 479 lines cut ...]
</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>A z2m external converter has been made.  https://gist.github.com/9zigen/34a27dd18dc4d4cd09c7b1a3efda8df4</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

https://gist.github.com/9zigen/34a27dd18dc4d4cd09c7b1a3efda8df4

9zigen, 2024-01-07:
Use this quirks 
https://gist.github.com/9zigen/3ec959af5b83722563f09826cef0021e

frootblaster, 2024-01-08:
Quirk isn't being applied.  Same values showing for CO2 as well.

9zigen, 2024-01-08:
Should work. 
Here's an example. Check your folder path with zha quirks in configuration.yaml:

```
zha:
  enable_quirks: true
  custom_quirks_path: /config/zha_quirks
```

[image]

frootblaster, 2024-01-08:
[image]
quirk_applied is still "false" despite settings:
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 9857,
    "manufacturer": "Alab",
    "model": "Alab-CO2-1.1",
    "name": "Alab Alab-CO2-1.1",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "quirk_id": null,
    "manufacturer_code": 4660,
    "power_source": "Mains",
    "lqi": 192,
    "rssi": -52,
    "last_seen": "2024-01-08T19:18:08",
    "available": true,
    "device_type": "Router",
    "signature": {

9zigen, 2024-01-08:
Do you have the core version of home assistant?

Try specifying the absolute path to the folder.
I think the problem is that the quirk is not loading.

if you start home assistant from console you should see in the log:
[image]

closed 2024-07-13 by github-actions[bot]