zigpy/zha-device-handlers #2891: Button "Push ME" from Marmitek

frank-schubidoo, 2024-01-08:
### Problem description

The button pairs just fine with ZHA but doesn't generate anything that can be used in an automation (onoff etc.). 

### Solution description

Very much a newbie here: guess a quirk is needed that makes that button be used in automations.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0006"
      ],
      "output_clusters": [
        "0x0006",
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TYZB01_1xktopx6",
  "model": "TS0041A",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0006"
      ],
      "output_clusters": [
        "0x0006",
        "0x000a",
        "0x0019"
      ]
    }
  },
[... 48 lines cut ...]
2024-01-08 14:28:10.688 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2024, 1, 8, 13, 28, 10, 688550, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0320), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=240, profile_id=260, cluster_id=6, data=Serialized[b'\x012\xfd\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=52, rssi=-87)
2024-01-08 14:28:10.690 DEBUG (MainThread) [zigpy.zcl] [0x0320:1:0x0006] Received ZCL frame: b'\x012\xfd\x01'
2024-01-08 14:28:10.690 DEBUG (MainThread) [zigpy.zcl] [0x0320:1:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=50, command_id=253, *direction=<Direction.Server_to_Client: 0>)
2024-01-08 14:28:10.691 WARNING (MainThread) [zigpy.zcl] [0x0320:1:0x0006] Unknown cluster command 253 b'\x01'
2024-01-08 14:28:10.692 DEBUG (MainThread) [zigpy.zcl] [0x0320:1:0x0006] Received command 0xFD (TSN 50): b'\x01'
2024-01-08 14:28:10.692 DEBUG (MainThread) [zigpy.zcl] [0x0320:1:0x0006] No explicit handler for cluster command 0xfd: b'\x01'
2024-01-08 14:28:10.692 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0x0320:1:0x0006]: received '0xFD' command with b'\x01' args on cluster_id '6' tsn '50'
2024-01-08 14:28:10.694 DEBUG (Thread-8) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0xffff789e3790>, 'UPDATE devices_v12\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_update_delta', {'ts': 1704720490.68855, 'ieee': 8c:f6:81:ff:fe:f4:92:d5, 'min_update_delta': 30.0})
2024-01-08 14:28:10.696 DEBUG (Thread-8) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0xffff789e3790>, 'UPDATE devices_v12\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_update_delta', {'ts': 1704720490.68855, 'ieee': 8c:f6:81:ff:fe:f4:92:d5, 'min_update_delta': 30.0}) completed
2024-01-08 14:28:10.698 DEBUG (Thread-8) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0xffff789e3790>)
2024-01-08 14:28:10.698 DEBUG (Thread-8) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0xffff789e3790>) completed

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

Eth030, 2024-01-26:
man i am so sad, just bought it ;(
[image]
it paired with skyconnect without issues but you cant use it for nothing

Eth030, 2024-01-26:
i found this quirk,
added a quirk folder to ha and added that path to config:
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0041.py
now it works

frank-schubidoo, 2024-01-29:
@Eth030 , how did you find that quirk and knew that this applies to this button? Just so next time I know where to look and what to do. thanks!

Eth030, 2024-01-29:
I’m sorry I don’t have a fact-based response. I just googled the device number, etc., and found a similar product and the quirk. I wasn’t sure if it would work, but it’s the same product, just differently branded. So, I tried, and it worked.

frank-schubidoo, 2024-01-29:
Nothing wrong with gut feeling. Thanks again!

closed 2024-08-03 by github-actions[bot]