zigpy/zha-device-handlers #2894: Add support for _TZE204_mhxn2jso variation of Tuya TXE mmWave sensor

tmcb82, 2024-01-08:
### Problem description

Add support for _TZE204_mhxn2jso variation of the Tuya mmWave sensor to the existing quirk.

### Solution description

Simple support can be added with `("_TZE204_mhxn2jso", "TS0601"),` under `class MmwRadarMotionGPP(CustomDevice):` I have tested it and this works. However like the other sensors in the class you have no control over changing settings or lux support.

### Screenshots/Video

[image]

### Device signature

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0107",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x000c",
        "0x0400",
        "0x0406",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_mhxn2jso",
  "model": "TS0601",
  "class": "tuya_ts0601_motion.MmwRadarMotionGPP"
}

danandrea737, 2024-02-25:
https://community.home-assistant.io/t/another-tuya-mmwave-human-presence-ts0601-tze204-mhxn2jso/681965/2

This is what I was able to achieve, but the sensor is still misbehaving. I have a clue this is related to the distance values being off.
Anyone has news about it?

Copyrighter, 2024-05-18:
any update on support?

closed 2024-11-21 by github-actions[bot]

twerff, 2025-04-23:
Please support this device please