zigpy/zha-device-handlers #2897: [Device Support Request] - TS011F by _TZ3000_okaz9tjs

arjanvdp, 2024-01-09:
### Problem description

I just received Tuya powerplugs 16A and 20A. 20A is detected with a quirk but the 16A, metering not showing the right values and some functions missing.

### Solution description

Trying to make my own local quirk but i dont get it to match the right way. Started with Plug_v2 from ts_011f_plug.py

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4660, maximum_buffer_size=108, maximum_incoming_transfer_size=0, server_mask=11264, maximum_outgoing_transfer_size=0, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000a",
        "0x0702",
        "0x0b04",
        "0x1000",
        "0xe000",
        "0xe001"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "_TZ3000_okaz9tjs",
  "model": "TS011F",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.1.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Amsterdam",
    "os_name": "Linux",
    "os_version": "6.1.63-haos-raspi",
    "supervisor": "2023.12.0",
[... 563 lines cut ...]
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    Identify.cluster_id,                    
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    LightLink.cluster_id,
                    Time.cluster_id,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                    TuyaZBMeteringClusterWithUnit,
                    TuyaZBElectricalMeasurement,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
```

</details>

### Additional information

When joining the device keep getting the mismatch: Fail because device_type mismatch on at least one endpoint

arjanvdp, 2024-01-09:
I have been messing around with the examples and did some reading.
This is my latest version... but not yet matched to the TS011F 16A Plug.

```
class Plug_16A(EnchantedDevice):
    """Another TS011F Tuya plug. First one using this definition is _TZ3000_okaz9tjs."""

    quirk_id = TUYA_PLUG_ONOFF

    signature = {
        MODEL: "TS011F",
        ENDPOINTS: {
            # "profile_id": 260, 
            # "device_type": "0x0100", 
            # "in_clusters": ["0x0000","0x0003","0x0004","0x0005","0x0006","0x000a","0x0702","0x0b04","0x1000","0xe000","0xe001"], 
            # "out_clusters": []
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Time.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    LightLink.cluster_id,                    
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    Time.cluster_id,
                    TuyaZBMeteringClusterWithUnit,
                    TuyaZBElectricalMeasurement,
                    LightLink.cluster_id,                    
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
```

hvf81, 2024-01-14:
Thank you for sharing the custom quirk.
I received the same type of powerplugs.
With this quirk they are working correctly Without it the power measurement did not work (way too high values)

closed 2024-07-28 by github-actions[bot]