zigpy/zha-device-handlers #2898: [Device Support Request]

dbaje, 2024-01-09:
### Problem description

Trying to add the Ikea Parasoll sensor in  ZHA results on a successful interview but it gets stuck at the configuring phase. Sometimes I get a few entities but they do not get updated and gets stuck on the value that was during pairing.

The same sensor was tested with Z2M and exact same hardware (home assistant instance was stopped and a Z2M instance was started using the same dongle) and it works there. This tell me  the support for this device is currently missing from ZHA.

### Solution description

Trying to add the Parasoll device should result of a successful pairing (and configuration) and then be able to use the sensor.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0103",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0b05",
        "0x1000",
        "0xfc7c",
        "0xfc81"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0019",
        "0x1000"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0500"
      ],
      "output_clusters": [
        "0x0003"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "PARASOLL Door/Window Sensor",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

[... 501 lines cut ...]
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0003": {
            "endpoint_attribute": "identify",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

[home-assistant_zha_2024-01-09T19-15-31.820Z.log](https://github.com/zigpy/zha-device-handlers/files/13878034/home-assistant_zha_2024-01-09T19-15-31.820Z.log)

</details>

MattWestb, 2024-01-09:
You is running on one TI coordinator i think and they was / is having some problems with Zigbee 3 end devices.
With EZSP is the device working great and is not having any connecting problem then being connected to one good router and not direct to the coordinator (that can being offline some times and end device is getting problems).

dbaje, 2024-01-09:
This can be closed. The solution of my problem was to add the device via another device in my zigbee network.

closed 2024-01-09 by dbaje

MattWestb, 2024-01-10:
Try updating the coordinator firmware then some of the Zigbee 3 functionality (but not all) was restored for one half year ago but i dont knowing if its working 100% OK.