zigpy/zha-device-handlers #2905: TZ3000_okaz9tjs shows me over 400 ampere as ative current

Mab306, 2024-01-13:
### Problem description

I have buy a zigbee plug from ali (TS011F
From _TZ3000_okaz9tjs). I could it pair it with zha but its Shows me unrealistic current in the energy monitoring [ more as 100 Amps] and i coulnt set up child switch and no set up backlight. Have some one the right quriks for that typ of tuya?
[image]
[image]

### Solution description

Right zha quirk for Show right current Amps and to set cild value and last state

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

Wittetonny, 2024-01-16:
Yes same problem, also interested in the right quirk for this power plug.

TheJulianJES, 2024-01-19:
You need to provide the device signature and/or diagnostics information for that device. It can be found on the device page.

Mab306, 2024-01-19:
I switch to zigbee2mqtt, device works there. Please zha u have great Potential but some Details work on zigbee2mqtt are really better but zha is easier.

closed 2024-01-19 by Mab306

Commifreak, 2024-02-25:
> You need to provide the device signature and/or diagnostics information for that device. It can be found on the device page.

I believe its the same signature as #3001

GuidoC1, 2024-03-08:
For others that might be looking for the same quirk:
I have the same device and applied the quirk provided in [this post](https://github.com/zigpy/zha-device-handlers/issues/1764#issuecomment-1946842958). It works fine for me and multiple other users. Hope this helps