zigpy/zha-device-handlers #2907: Aqara H1EU Double Rocker WS-EUK04 - some exposed entities/sensors are incorrect/not working [BUG]

danielshiloah, 2024-01-14:
### Bug description

Recently new support for Aqara WS-EUK04 in ZHA was added by @TheJulianJES. much appreciated btw!
Using ZHA on Core 2024.1.3 , Supervisor: 2023.12.0, Operating System: 11.4, Frontend: 20240104.0

I found a few bugs in the exposed entities:
1. Both switches are labeled the same as Switch (not switch 1/2 or L/R) I changed the name of one switch to switch2.
2. The "Summation delivered" sensor does not work, state is always 0 (left lights on for a hole day)
3. The "Voltage" sensor does not work, always on 0V (in my case it should be 220V)

[image]

### Steps to reproduce

Add WS-EUK04 to ZHA via coordinator.
Look at added device in HA devices list.  

### Expected behavior

Expect each rocker(switch) to have a different name , ex: Switch1, Switch 2 or Switch L, Switch R.
Expect "Summation delivered" sensor to show the sum of all power used.
Expect "Voltage" sensor to show correct voltage, ex: 220V or 110V. 

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Additional information

Was asked by @TheJulianJES (https://github.com/TheJulianJES) to open issue.

thedeemling, 2024-02-10:
Also 4) The `switch_mode` attribute does not work and always returns `None`:
[image]

This attribute disables the 500ms waiting period for events - enabling instantaneous single-click-only mode.

andrewolfy, 2024-02-24:
Hi all,

I have home assistant Yellow
Core 2024.2.3
Supervisor 2024.02.0
Operating System 11.5
Frontend 20240207.1

and I'm experience this dificultie for Aqara WS-EUK04 and also to Aqara WS-EUK03

[image]

[image]

As this solution been implemented in the new release?

baulaergrungim, 2024-05-26:
I have the same issue on Aqara H1 double rocker with neutral 
[image]
As per suggestion on Discord, I tiried reconfiguring it, and even removing and readding it, didn't help. I use SkyConnect's dongle and RPi5 in Supervisor as my HA server.

homerojleal, 2024-11-06:
I have a couple of these with same issue (single/double rockers).
Lastest version of HA and coordinator is SkyConnect v1.0 Firmware: 7.4.3.0 build 0
Added/removed one H1 to se if issue was resolved, but no luck.

lanfeustdsn, 2025-01-27:
Same here  - summation and voltage only report 0

[image]

baulaergrungim, 2025-03-22:
When I was bored, I tried fixing it with a helper. Adding this is a good workaround https://www.home-assistant.io/integrations/integration/ I used Left Riemann with max 1min non-reporting, and it even shows on the switch overview page (I think you need to name it something different than the existing entity name).

Cyruz143, 2025-09-19:
Still an issue with 2025.9

[image]