zigpy/zha-device-handlers #2909: [Device Support Request] Schneider CCT5015-0002W

Pierrou25, 2024-01-14:
### Problem description

The module is recognized and up and down functions are working but no timing solutions exists. Could you provide timer function? 

### Solution description

Ah timer setting function

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

TheJulianJES, 2024-01-19:
You need to provide the device signature and/or diagnostics information for that device. It can be found on the device page.

Pierrou25, 2024-01-20:
Please find the device signature below

LIAISONS

SIGNATURE

VOISINS
1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
16
17
18
19
20
21
22
23
24
25
26
27
28
29
30
31
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4190, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "5": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0102",
        "0x0b05"
      ],
      "output_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "Schneider Electric",
  "model": "PUCK/SHUTTER/1",
  "class": "zigpy.device.Device"
}

Best regards

luke7101, 2024-03-12:
What's a timing function ?
(en FR si tu veux)

Pierrou25, 2024-03-12:
Merci Luke.
Voici plus d'info sur le problème rencontré.
ce module est bien reconnu sous ZHA et les fonctions montee descente du volet fonctionne mais il n'est pas possible de définir la durée de montée descente donc le volet n'est pas calibré. Apparemment cette fonction existe sur zigbee 2mqtt. Merci de votre aide.

Pierrou25, 2024-03-16:
Problem solved with switching from zha to zigbee2mqtt. All functions are available.

closed 2024-03-16 by Pierrou25