zigpy/zha-device-handlers #2914: [Device Support Request] Danfoss Icon2 Room Sensor (088U2121)

JohnConnett, 2024-01-16:
### Problem description

I would like to have the [Danfoss Icon2 Room Sensor (088U2121)](https://assets.danfoss.com/documents/354092/AI380762595805en-010301.pdf) fully supported under ZHA.

Note that there is a similar request for zigbee2mqtt, [Danfoss Icon2 Room Sensor (088U2121)](https://github.com/Koenkk/zigbee2mqtt/issues/20222). 

The device is recognized by ZHA but some values returned are incorrect:

**Device info** 0x8040 by Danfoss Firmware: 0x0000010c
IEEE: 8c:f6:81:ff:fe:82:4a:00
Nwk: 0xc938
Device Type: EndDevice
LQI: 255
RSSI: -33
Last seen: 2024-01-15T16:33:34
Power source: Battery or Unknown
**Controls**
Thermostat Heat 21 °C Currently: 24.2 °C
**Sensors**
Humidity 26.0%
HVAC action Unknown
Temperature -327.7 °C
Diagnostic
Battery 100%
Identity PRESS
LQI
RSSI
**Logbook**
January 15, 2024
_Danfoss 0x8040 Identify_ changed to January 15, 2024 at 4:20 PM triggered by service Button: Press
4:20:48 PM - 25 minutes ago - John Connett

Note that the **Thermostat** temperature is correct (24.2 °C) but the **Sensors** temperature is wrong (-327.7 °C) 

### Solution description

I would like the **Danfoss Icon2 Room Sensor (088U2121)** to be fully supported.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4678, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0301",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0201",
        "0x0204",
        "0x0402",
        "0x0405",
        "0x0b05"
      ],
      "output_clusters": [
        "0x0003",
[... 462 lines cut ...]
}

```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

JohnConnett, 2024-01-16:
As noted against the zigbee2mqtt issue:

According to the **Cluster Library Specification** Document 07-5123 Revision 8 in section 4.4.2.2.1 _MeasuredValue_ Attribute: A _MeasuredValue_ of 0x8000 indicates that the temperature measurement is unknown.

So it appears to be a valid response when the temperature measurement is unknown. Perhaps there should be a test for that specific value rather than reporting -327.68 °C?

JohnConnett, 2024-01-17:
This may be slightly off-topic. I was comparing the contents of **Download diagnostics** with attributes visible through **Manage zigbee device**.

For example, consider the **Cluster Library Specification** Document 07-5123 Revision 8, section **6.6 Thermostat User Interface Configuration**. The relevant section of the diagnostics is:
```
          "0x0204": {
            "endpoint_attribute": "thermostat_ui",
            "attributes": {},
            "unsupported_attributes": {}
```
However, if inspected using **Manage zigbee device** then three attributes are visible, two mandatory: _TemperatureDisplayMode_ (Id: 0x0000) and _KeypadLockout_ (Id: 0x0001); and one optional: _ScheduleProgrammingVisibility_ (Id: 0x0002). READ ATTRIBUTE returns values for the two mandatory attributes and `None` for the optional attribute.

I would have expected **Download diagnostics**, in this case, to list the two mandatory attributes and list _ScheduleProgrammingVisibility_ in unsupported_attributes. I'm not sure if this is a bug or a feature ...

My experience with ZHA is very limited. If there is a tool that can characterize which attributes are supported by a new device that would be very useful. Would stepping through all the options in **Manage zigbee device** uncover them?

JohnConnett, 2024-07-16:
Still interested in full support for this device. Will put in more effort if time permits.

kongsted, 2024-10-04:
If I can be at any help with getting info about the device (we're having both 088U2121 and 088U2122 in our house), I would gladly help.
Unfortunately my knowledge on how to create quirks and similar in HA is limited (haven't done that yet).
//Anders

closed 2025-08-15 by github-actions[bot]