zigpy/zha-device-handlers #2917: [Device Support Request] Too many sensors on _TZ3000_zw7yf6yk

phulstaert, 2024-01-17:
### Problem description

Although we do it generally the other way around, I have too many sensors.

I have a _TZ3000_zw7yf6yk ([zigbee 3.0 switch](https://zigbee.blakadder.com/Girier_JR-ZDS01.html)).
It is very simple: you can turn it on... and even off.
But when adding it, it shows all different kind of sensors (Current, Power, Summation delivered and Voltage). This device does not do any form of energy monitoring and the values remain on 0.

### Solution description

I think it would be better to only expose the sensors or options that are available on the device. 
I have not found a _TZ3000_zw7yf6yk that does have energy monitoring.
I don't know about energy monitoring on _TZ3000_majwnphg or _TZ3000_6axxqqi2 (from [here](https://zigbee.blakadder.com/Girier_JR-ZDS01.html)) so i would leave them untouched for now...

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

TheJulianJES, 2024-01-19:
You need to provide the device signature and/or diagnostics information for that device, so we know what quirk it currently matches. It can be found on the device page.

You can also hide or disable sensors/entities in HA.

gvillo, 2024-07-08:
@TheJulianJES does this info works for you? I am experiencing the same as @phulstaert, power measurement is not working at all, AFAIK this device does support that.

Device: _TZ3000_zw7yf6yk (https://zigbee.blakadder.com/Girier_JR-ZDS01.html)
Quirk applied: zhaquirks.tuya.ts000x.Switch_1G_Metering

`json
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 39565,
    "manufacturer": "_TZ3000_zw7yf6yk",
    "model": "TS0001",
    "name": "_TZ3000_zw7yf6yk TS0001",
    "quirk_applied": true,
    "quirk_class": "zhaquirks.tuya.ts000x.Switch_1G_Metering",
    "quirk_id": "tuya.plug_on_off_attributes",
    "manufacturer_code": 4417,
    "power_source": "Mains",
    "lqi": 255,
    "rssi": -73,
    "last_seen": "2024-07-07T23:31:56",
    "available": true,
    "device_type": "Router",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0100",
          "input_clusters": [
            "0x0000",
            "0x0003",
            "0x0004",
            "0x0005",
            "0x0006",
            "0x0702",
            "0x0b04",
            "0xe000",
            "0xe001"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        }
      },
      "manufacturer": "_TZ3000_zw7yf6yk",
      "model": "TS0001"
    },
    "active_coordinator": false,
    "entities": [
      {
        "entity_id": "button.switch_test_identify",
        "name": "_TZ3000_zw7yf6yk TS0001"
      },
      {
        "entity_id": "light.switch_test_light",
        "name": "_TZ3000_zw7yf6yk TS0001"
      },
      {
        "entity_id": "select.switch_test_power_on_state",
        "name": "_TZ3000_zw7yf6yk TS0001"
      },
      {
        "entity_id": "select.switch_test_backlight_mode",
        "name": "_TZ3000_zw7yf6yk TS0001"
      },
      {
        "entity_id": "sensor.switch_test_current",
        "name": "_TZ3000_zw7yf6yk TS0001"
      },
[... 254 lines cut ...]
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {
              "0x0002": {
                "attribute_name": "current_file_version",
                "value": 72
              }
            },
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
`

phulstaert, 2024-07-08:
The TZ3000_zw7yf6yk does not support energy monitoring. (cfr https://zigbee.blakadder.com/Girier_JR-ZDS01.html)
Also, when you look at the guts of the device, you will see there is no current sensor; just a simple relay.

closed 2025-01-11 by github-actions[bot]

wise-rice, 2025-02-10:
Can this please be reopened? I have the same device.

Also: The Quirk is missing a key function: The device can function as a rocker switch or momentary. This is a crucial setting and it is missing. 
Z2m has it.