zigpy/zha-device-handlers #2919: EBYTE Zigbee to RS 485 Modbus E180-DTU(Z20-485)

Jusch777, 2024-01-18:
### Problem description

I purchased a device from EBYTE which should convert RS485 registers to Homeassistant.
Registers are read and write with function 03 and 10
Want to use this device to connect my EV Wallbox which communicates only serial via RS485 .
Connection is in Garage, so need Zigbee network to get over a distance of 20 m.

Issue: Homeassistant pairs with the device, but only 3 useless entities are provided: 
LQI
RSSI
Identify

The Wallbox requires to read and write Register 1000 - 1010 and 2000-2010
Read function is 03
Write function is 10 (write multiple registers)

### Solution description

There is a documeentation of the device communication. But I am not able to find out how I can create Entities in Homeassistant for the entities of the certain registers.

### Screenshots/Video

Documentation of the device:
(https://www.cdebyte.com/products/E180-DTU(Z20-485)

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

TheJulianJES, 2024-01-19:
You need to provide the device signature and/or diagnostics information for that device. It can be found on the device page.

Jusch777, 2024-01-22:
[zha-81d2b1300f276c0085046bb1f82b0c4b-EBYTE FW7424-3-11-a4e157a3d626ea43adae0fffce169f16.json(1).txt](https://github.com/zigpy/zha-device-handlers/files/14015942/zha-81d2b1300f276c0085046bb1f82b0c4b-EBYTE.FW7424-3-11-a4e157a3d626ea43adae0fffce169f16.json.1.txt)

Hi, file attached. Thanks !

jpcool23, 2024-02-11:
Hi guys, im new to this HA stuff and i try to read out my solis inverter via the same EBYTE-Adapter
I have already linked it to my ZHA-network but there is no data comming in. Could u please explain how u get it to work ? 
thank u =)

closed 2024-08-16 by github-actions[bot]