zigpy/zha-device-handlers #2921: [Device Support Request] ACOVA Alcantara heater support in ZHA

stadros83, 2024-01-19:
### Problem description

Hello

I just bought a zigbee heater, the Acova Alcantara.
I can pair it to ZHA and get entities but there is a "bug".

If I change the temperature using ZHA, the heater stop being in "box mode" (so connected to ZHA) and goes to idle mode.

I'm not the only one with this issue : https://community.home-assistant.io/t/acova-alcantara-radiator-with-install-code/469444

It seems to work well with Z2M, but I'm using ZHA and don't want to migrate everyting :(
I'm not a developper but I can provide logs or whatever is needed :) !

### Solution description

I would like to be able to change the temperature of the heater without losing connection to it.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0301",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0201"
      ],
      "output_clusters": [
        "0x0003"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0107",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0406"
      ],
      "output_clusters": [
        "0x0003"
      ]
    },
    "3": {
      "profile_id": "0x0104",
      "device_type": "0x0001",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x000f"
      ],
      "output_clusters": [
        "0x0003"
      ]
    },
[... 559 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

stadros83, 2024-01-21:
I've seen in the developpers tools, on climate.radiateur_NAME_thermostat : 

If I have this it is NOT working : 
system_mode: "[<SystemMode.Auto: 1>]/heat_cool"

If I have this it IS working : 
system_mode: "[<SystemMode.Emergency_Heating: 1>]/heat"

stadros83, 2024-01-22:
It seems that the problem is coming from SystemMode.AUTO.
For HA Auto means that the heater is controled by a schedule, but for this manufacturer AUTO means that the heater is in box mode and can receive commands.

stadros83, 2024-01-24:
I want to add that I can control the heater from ZHA by changing the value of occupied_heating_setpoint.

Example : 

  - service: zha.set_zigbee_cluster_attribute
    data:
      cluster_type: in
      ieee: xx:xx:xx
      endpoint_id: 1
      cluster_id: 513
      attribute: 18
      value: "1200"   // value is in celsius * 100, so here 12 degres

stadros83, 2024-02-05:
No one :( ?

gdoffe, 2024-09-04:
> It seems that the problem is coming from SystemMode.AUTO. For HA Auto means that the heater is controled by a schedule, but for this manufacturer AUTO means that the heater is in box mode and can receive commands.

I re-up this topic as I have a Taffetas 2 that seems to have a similar behavior.

I'm quite new in ZHA development and just discovered quirks. It seems to me that quirks are useful to fix raw data coming from devices according to their clusters.

However here it is more the functional behavior of the heater itself that is problematic.
As you mention SystemMode.Auto means you can control the heater through Zigbee (heating only).
Switching to SystemMode.Heat moves the heater into a Manual mode.

So the only two interesting system modes from ZHA point of view are:
- SystemMode.Auto
- SystemMode.Off

So SystemMode.Auto is not corresponding to HVACMode.HEAT_COOL but to HVACMode.HEAT.

So ZHA conversion tables defined as constants in [application/platforms/climate/const.py](https://github.com/zigpy/zha/blob/c6edfd47997f2023427514d558da90a0ecf75688/zha/application/platforms/climate/const.py#L168) are not applicable here and must be redefined:

```python
  HVAC_MODE_2_SYSTEM = {                                                          
      HVACMode.OFF: SystemMode.Off,                                               
      HVACMode.HEAT_COOL: SystemMode.Auto,                                        
      HVACMode.COOL: SystemMode.Cool,                                             
      HVACMode.HEAT: SystemMode.Heat,                                             
      HVACMode.FAN_ONLY: SystemMode.Fan_only,                                     
      HVACMode.DRY: SystemMode.Dry,                                               
  }                                                                               
                                                                                  
  SYSTEM_MODE_2_HVAC = {                                                          
      SystemMode.Off: HVACMode.OFF,                                               
      SystemMode.Auto: HVACMode.HEAT_COOL,                                        
      SystemMode.Cool: HVACMode.COOL,                                             
      SystemMode.Heat: HVACMode.HEAT,                                             
      SystemMode.Emergency_Heating: HVACMode.HEAT,                                
      SystemMode.Pre_cooling: HVACMode.COOL,  # this is 'precooling'. is it the same?
      SystemMode.Fan_only: HVACMode.FAN_ONLY,                                     
      SystemMode.Dry: HVACMode.DRY,                                               
      SystemMode.Sleep: HVACMode.OFF,                                             
  } 
```
  
For me it seems more related to ZHA integration, so it could not be done inside a quirks, leading to the following patch on zha:
```diff
diff --git a/zha/application/platforms/climate/__init__.py b/zha/application/platforms/climate/__init__.py
index a297a43..37dfc64 100644
--- a/zha/application/platforms/climate/__init__.py
+++ b/zha/application/platforms/climate/__init__.py
@@ -581,6 +581,75 @@ class ZenWithinThermostat(Thermostat):
     """Zen Within Thermostat implementation."""
 
 
+@MULTI_MATCH(
+    cluster_handler_names=CLUSTER_HANDLER_THERMOSTAT,
+    manufacturers={"ZEHNDER GROUP VAUX ANDIGNY      "},
+    stop_on_match_group=CLUSTER_HANDLER_THERMOSTAT,
+)
+class ZehnderThermostat(Thermostat):
+    """Zehnder thermostat to adapt AUTO mode behavior."""
+    ZEHNDER_HVAC_MODE_2_SYSTEM = {
+        HVACMode.OFF: SystemMode.Off,
+        HVACMode.HEAT: SystemMode.Auto,
+    }
+
+    ZEHNDER_SYSTEM_MODE_2_HVAC = {
+        SystemMode.Off: HVACMode.OFF,
+        SystemMode.Auto: HVACMode.HEAT,
+        SystemMode.Heat: HVACMode.HEAT,
+    }
+
+    async def async_set_hvac_mode(self, hvac_mode: HVACMode) -> None:
[... 31 lines cut ...]
+            f"[{thermostat.system_mode}]/{system_mode}"
+            if self.hvac_mode is not None
+            else None
+        )
+        response[ATTR_OCCUPANCY] = thermostat.occupancy
+        response[ATTR_OCCP_COOL_SETPT] = thermostat.occupied_cooling_setpoint
+        response[ATTR_OCCP_HEAT_SETPT] = thermostat.occupied_heating_setpoint
+        response[ATTR_PI_HEATING_DEMAND] = thermostat.pi_heating_demand
+        response[ATTR_PI_COOLING_DEMAND] = thermostat.pi_cooling_demand
+        response[ATTR_UNOCCP_COOL_SETPT] = thermostat.unoccupied_cooling_setpoint
+        response[ATTR_UNOCCP_HEAT_SETPT] = thermostat.unoccupied_heating_setpoint
+        return response
+
+    @property
+    def hvac_mode(self) -> HVACMode | None:
+        """Return HVAC operation mode."""
+        return ZehnderThermostat.ZEHNDER_SYSTEM_MODE_2_HVAC.get(self._thermostat_cluster_handler.system_mode)
+
+
 @MULTI_MATCH(
     cluster_handler_names=CLUSTER_HANDLER_THERMOSTAT,
     aux_cluster_handlers=CLUSTER_HANDLER_FAN,
```

As I am completly noob to this, does it seem correct ?

gdoffe, 2024-09-04:
@stadros83 the issue has been automatically closed, maybe you could re-open it ?

stadros83, 2024-09-04:
Not sure it is possible to reopen (at least I probably don't have the rights to do that).

gdoffe, 2024-09-04:
Thank you @dmulcahey !

@stadros83 could you test on your side ?

stadros83, 2024-09-05:
Test what exactly ?