zigpy/zha-device-handlers #2923: Heiman SmokeSensor-EM - yet another HS1SA (only drop wrongly reported IAS WD cluster)

masi, 2024-01-19:
### Problem description

The device Heiman Smoke Sensor HS1SA reported as SmokeSensor-EM shows a IAS WD cluster, but it isn't an intruder alarm system.

The device works properly (smoke is detected and reported) but in the ZHA UI the (configuration) controls don't work.

### Solution description

```

class HeimanSmokeEM(CustomDevice):
    """SmokeEM quirk."""

# NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0,
# reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>,
# mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4619, maximum_buffer_size=82,
# maximum_incoming_transfer_size=82, server_mask=0, maximum_outgoing_transfer_size=82,
# descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True,
# *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False,
# *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
    signature = {
        MODELS_INFO: [("HEIMAN", "SmokeSensor-EM")],
        ENDPOINTS: {
            # "profile_id": "0x0104", "device_type": "0x0402",
            # "input_clusters": ["0x0000", "0x0001", "0x0003", "0x0500", "0x0502"],
            #  "output_clusters": ["0x0019"]
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    IasZone.cluster_id,
                    IasWd.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                ],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    IasZone.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                ],
            },
        },
    }
```

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

[... 464 lines cut ...]
                ],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    IasZone.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                ],
            },
        },
    }
```

</details>

TheJulianJES, 2024-01-19:
So also just removing the `IasWd` cluster (like for the other sensors).

Do you want to create a PR for this or should I PR that change quickly?

masi, 2024-01-19:
> So also just removing the `IasWd` cluster (like for the other sensors).

That's the idea.

> Do you want to create a PR for this or should I PR that change quickly?

I created my quirk by copying the quirk class HeimanSmokeEF30, but that's it. Shouldn't I test the quirk?

Ie copy the imports and my draft an put it there:
```

zha:
  custom_quirks_path: /config/custom_zha_quirks/

```

TheJulianJES, 2024-01-19:
Ah, yeah. You should be able to copy the full py file with (all) Heiman quirks into that folder, restart HA, and see if it works.

closed 2024-01-27 by TheJulianJES