zigpy/zha-device-handlers #2926: [Device Support Request] TS0601 _TZE200_icka1clh  {MOES AM43}

darknessblade, 2024-01-20:
### Problem description

device is reconized as a Temperature sensor instead of a "Cover/blinds controller"

Link to order page:
https://www.aliexpress.com/item/1005005447997612.html?

### Solution description

Custom quirk, or adding the device ID/manufactuer information to the correct ZHA entity

### Screenshots/Video

[image]

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

darknessblade, 2024-01-20:
Checking the quirks, using it trough a custom quirk the following one looks to match
"class TuyaMoesCover0601(TuyaWindowCover):"

Only thing it doesn't transmit is Battery information.

Berenhappinda, 2024-01-24:
Problem seems to be the same as: https://github.com/zigpy/zha-device-handlers/issues/2936

nikolaynikolov681, 2024-04-09:
Hey, I have same issue :(

closed 2024-10-13 by github-actions[bot]