zigpy/zha-device-handlers #2929: [Device Support Request] Tuya Zigbee Temperature Humidity Sensor

ItamarVa, 2024-01-21:
### Problem description

I managed to add the ZB device but it is missing the main sensors (temp and humidity)
https://www.aliexpress.com/item/1005005234573063.html?spm=a2g0o.order_list.order_list_main.48.66d01802bShOVQ
[image]

### Solution description

add the relevant sensors of temperature and humidity to the device

### Screenshots/Video

<details><summary>Screenshots/Video</summary>
[image]

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_9yapgbuv",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.1.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Asia/Jerusalem",
    "os_name": "Linux",
    "os_version": "6.1.71-haos",
    "supervisor": "2023.12.1",
    "host_os": "Home Assistant OS 11.4",
[... 335 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

dualmacops, 2024-02-13:
Model name is: ZTH02

alucar-d, 2024-03-02:
I have another one if you need testing

delaineS, 2024-03-03:
Have u found a solution?

alucar-d, 2024-03-03:
Actually I'm using the [last Tuya sensor quirk](https://github.com/zigpy/zha-device-handlers/blob/e4d0663aa4fb2f4bd0b41825f1942e3d29893375/zhaquirks/tuya/ts0601_sensor.py). 
I added the device model `("_TZE204_9yapgbuv", "TS0601")` to the `TuyaTempHumiditySensorVar04` and after repair the device and wait some time the data appear!
I need to compare the values with other similar sensor I have, but it looks good. I got the device some days ago and the battery show 25%, but maybe it's normal for a device from aliexpress.

ItamarVa, 2024-03-05:
> Actually I'm using the [last Tuya sensor quirk](https://github.com/zigpy/zha-device-handlers/blob/e4d0663aa4fb2f4bd0b41825f1942e3d29893375/zhaquirks/tuya/ts0601_sensor.py). I added the device model [...]

it works! finally I get some info from it..... thank you!!

delaineS, 2024-03-05:
I tried, at first everything worked perfectly then after a while it went offline and no longer wants to know if it's working

agvg1, 2024-08-17:
Have this fix been added to the official release?

PR #3817 (2025-01-31): Fix Tuya temp/humidity sensor quirks (`_TZE200_upagmta9`, ...)
## Proposed change

Add Tuya `tze200 cirvgep4` and combine quirks to match z2m.

See: https://github.com/Koenkk/zigbee-herdsman-converters/blob/09cfce5153d3931c7cc009aa3ae3107d9b394f5c/src/devices/tuya.ts#L1073

## Additional information

Closes: #2577
Closes: #2854 
Closes: #2304 
Closes: #3260 
Closes: #2725 
Closes: #2929 

## Checklist

- [ ] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [x] Tests have been added to verify that the new code works
closed 2025-01-31 by TheJulianJES