zigpy/zha-device-handlers #2931: [Device Support Request] “TS011F by _TZ3000_cayepv1a“ Tongou Variable Load Relay Switch

npapagiannis, 2024-01-22:
### Problem description

The zha integration does not support the variable load (1-63A) feature (as far as I know it does work on z2m)

### Solution description

I hope the gurus can come up with a working quirk

### Screenshots/Video

<details><summary>Screenshots/Video</summary>
(I’ve unplugged the device, that’s why it shows as unavailable)
[image]
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x010a",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0402",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_cayepv1a",
  "model": "TS011F",
  "class": "zigpy.device.Device"
}

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

Xesc1714, 2024-02-03:
I not found any integration for this breaker/meter device..., HA found through ZHA, but it can't show better than TUYA.
Limit Ampers can't be change 1-63. or Leakeage value...  
TUYA are still light years away from any. Continous use it with app TUYA (also easy & friendly installation) .
[image]
Any way thank you.

pulento, 2024-05-26:
Found the commit con zigbee2mqtt that added support for this devices with thresholds here

[https://github.com/Koenkk/zigbee-herdsman-converters/commit/c9f531f50a7c819050ddb9cc20d21367288afa1c](url)

franortiz, 2024-10-03:
I created a quirks v2 for this model, working great with all the breakers and thresholds

https://gist.github.com/franortiz/3fd91c6fa0884f2e33fa7e9d57ad0930

pulento, 2025-03-04:
> I created a quirks v2 for this model, working great with all the breakers and thresholds
> 
> [quirk v2](https://gist.github.com/franortiz/3fd91c6fa0884f2e33fa7e9d57ad0930)

I installed on my Hass.io it shows the thresolds but if I change them and then reload the device info page it comes to defaults for example "Overcurrent Threshold" come back to 32. And if I chedk the switch Overcurrent Breaker it automatically comes back to OFF.

[image]

pulento, 2025-03-06:
Ahhhh I didn't remove the device. I'm waiting for a new one (since now its my main switch I cannot fiddle with it) and report back if after re-adding it everything works finde.

How do I activate ZHA in debug mode ? through the YAML configuration I suppose.

Thanks a lot.

franortiz, 2025-03-08:
> [@franortiz](https://github.com/franortiz)
> 
> I did re-pair my device and the thresholds are workin fine ! Thanks a lot !

Vamooo

closed 2025-09-13 by github-actions[bot]