zigpy/zha-device-handlers #2932: [Device Support Request] Iris Security Keypad (v3) Model: iL02_1

consolecwby, 2024-01-22:
### Problem description

I have this device in Hubitat and it reports the following states:
Current States
alarm : off
armingIn : 0
battery : 100
codeChanged : added
codeLength : 4
lastCodeName : code #1
lockCodes : {"1":{"name":"code #1","code":"2217"}}
maxCodes : 20
motion : inactive
securityKeypad : disarmed
tamper : clear
temperature : 76.82

On Home assistant using default setup the device only gives the following:
Controls:
Alarm control panel
Disarmed
Sensors:
Humidity: 30.8 %
IAS zone: Off
Temperature: 78.1 °F
Diagnostic
Battery: Unknown
Identify
LQI
RSSI

### Solution description

Would like to be able to control pin input, Battery status worked, control arming delay time and other features: would like to be able to sync with alarmo

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4667, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0402",
        "0x0405",
        "0x0500",
        "0x0501",
        "0x0b05",
        "0xfc01",
        "0xfc02"
      ],
      "output_clusters": [
        "0x0003",
        "0x0019",
[... 484 lines cut ...]
### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

The hubitat driver they use can be found here.

https://github.com/hubitat/HubitatPublic/blob/master/examples/drivers/irisKeypadV3.groovy

closed 2024-07-27 by github-actions[bot]

consolecwby, 2024-07-28:
boo

werddomain, 2024-08-13:
😥