zigpy/zha-device-handlers #2933: [Device Support Request] Tuya Zigbee Smart Curtain Robot _TZE200_cpbo62rn

david-romero, 2024-01-23:
### Problem description

the device is integrated without any sensor or control

### Solution description

- The device has controls for closing or opening like a cover
- The device has a sensor with the battery
- The device has a sensor with the luminosity 

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0102",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_cpbo62rn",
  "model": "TS0601",
  "class": "ts0601_cover.TuyaMoesCover0601"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.1.5",
[... 745 lines cut ...]
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerWindowCover,
                    TuyaWindowCoverControl,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
        }
    }
```

</details>

i8nemo, 2024-02-08:
i also have these units

BlikeBread, 2024-03-13:
Thanks for sharing it. I added the quirk and I can now control the open/close/stop function. however, i do not see any sensors like battery and luminosity. Any suggestions?

GuidoC1, 2024-03-15:
> Thanks for sharing it. I added the quirk and I can now control the open/close/stop function. however, i do not see any sensors like battery and luminosity. Any suggestions?

Nope, I have not tried to dive into it myself yet. I also have no experience yet, as I have never tried to debug zigbee devices in order to figure out what to add to the custom quick.

P.S.
The missing sensors (battery, luminosity, calibration settings), that is exactly what this device support request is for. Hence it is still open :)

SebastianBrase85, 2024-06-09:
I'd also love to have this

lukechadwick, 2024-09-19:
This worked well with these:
https://www.aliexpress.com/item/1005007045870483.html

I purchased two.  Only one of them seems to pair but controls both.   They both spin in the same direction so one needs to be flipped around. Not sure if they can be addressed individually but not really an issue as I have them hidden on the other side of the curtain.

Thanks!

Arnok666, 2024-10-23:
I also request, thank you

1liminal1, 2024-12-30:
Hey guys, 
How can you set open and closed limits?

Arnok666, 2024-12-30:
I actually just guessed and trial and errored it travelling the curtain rod.
50% was all i needed to close my curtain!

GuidoC1, 2024-12-30:
> Hey guys, 
> How can you set open and closed limits?

If you have a remote for the device, then you can use the remote to set open/close limits.

P.S.
Setting limits with ZHA is not supported in the current custom quirk version. Hence the support request is still open.

1liminal1, 2024-12-31:
Oh ok, so useless then, until it gets supported? Thats a shame

Thanks :)

1liminal1, 2024-12-31:
Is this the only forum to get it supported? This ticket has been open for almost a year?

Thanks!

lukechadwick, 2024-12-31:
They seem to have some collision detection anyway, mine stop when they meet the centre rail hook.

prairiesnpr, 2024-12-31:
This quirk may work, it's untested but based off of the z2m converter.

https://github.com/prairiesnpr/zha-device-handlers/blob/convert-tuya-covers-v2/zhaquirks/tuya/tuya_cover_v2.py

1liminal1, 2025-01-02:
Hey thanks for this, I added this new quirk. But still no set limits or battery. 

Am I missing something? 

[image]

Thanks guys

prairiesnpr, 2025-01-02:
> Hey thanks for this, I added this new quirk. But still no set limits or battery.
> 
> Am I missing something?

Under Zigbee info, what quirk does it say is applied?

1liminal1, 2025-01-02:
It doesnt? 

It must be picking it up though, when I first added the device, It had no features so

[image]

prairiesnpr, 2025-01-03:
Have you tried removing it, enabling debug logging and then rejoining and see if anything shows up in the logs. It should show that it either matched the quirk or give an error.

1liminal1, 2025-01-04:
I can only get the Quirk at the top to do anything, Nothing else seems to work. Pretty crappy considering these things are $140 bucks, but what can you do if you're not a dev lol. 

I think Ill use a some input booleans and a timer in node red to open and close them, should work well enough. Costly mistake though :(

prairiesnpr, 2025-01-04:
@1liminal1 I had a copy / paste error on that quirk, please try it again.

1liminal1, 2025-01-05:
Thank you for that :) I just restarted and have the options, though battery is showing unknown. 

Ill test it later today and see if its working :) 

[image]

prairiesnpr, 2025-01-12:
@1liminal1 Did you have any success?

prairiesnpr, 2025-01-12:
I basically just copied what z2m did, I don't have one to test with. Look at the conversation here, https://github.com/Koenkk/zigbee2mqtt/issues/17436#issuecomment-1600901453 they have a sequence of commands. If it feels like something isn't quite right let me know.

1liminal1, 2025-01-13:
Thanks mate! I'm about to go OS, so not sure when I will get a chance to test. So if you don't hear from me for a while. Thanks for all this :)

1liminal1, 2025-03-05:
How did you go? I never got it working well. It also drops off a lot

GuidoC1, 2025-03-06:
> I can only get the Quirk at the top to do anything, Nothing else seems to work. Pretty crappy considering these things are $140 bucks, but what can you do if you're not a dev lol.
> 
> I think Ill use a some input booleans and a timer in node red to open and close them, should work well enough. Costly mistake though :(
> 
> One more thing, how does the remote work though? How does it pair to the devices?

@1liminal1 , Do you have a remote?
How it works, is something I would expect to be described in the manual of the remote/curtain bot.
I will attach the instructions that were included with the curtain bot sets that I use.

@1liminal1 , additionally (in case you have a remote):
Not sure if the following info is already known to you, nor if it is useful for your use-case, but thought I should share it just in case it would be helpful for anyone :)

To make the curtain bot & custom quirk "useful" for home automation, I did the following:
1) Install custom quirk
2) Connect curtain bot with ZHA
3) Use the remote (of the curtain bot) to set open/close limits.
Note: the by remote configured limits, will also be adhered to when using ZHA open/close/set % commands and buttons.
4) .
a) Use automations to open/close the curtains based on time, temperature, sunrise, sunset, solar azimuth/percentage, etc.
b) Use the Home Assistant GUI or curtain bot remote, to manually open/close the curtains.

The batteries of the curtain bot last ~6 months when opening & closing the curtains once a day. So the missing battery sensor would definitely be nice to have, but should not frequently be an issues.
The missing luminosity sensor could be useful for certain use cases, however, here you could simply use any other luminosity sensor in your automations. Of course a shame to add additional devices for something which another devices already contains (in hardware)..

Anyways, for the people who have a remote, the custom quick should be sufficient for most use cases. Hopefully also for you to not consider it "useless" & "a costly mistake"  :(
In case you don't have a remote, I have seen some vendors on AliExpress sell the remote separately for just a few bucks.

**Details of the referred to HW & SW:**
Hardware:
[AliExpress_link1](https://www.aliexpress.com/item/1005006080554175.html?spm=a2g0o.order_list.order_list_main.9.531a79d2WIOGie&gatewayAdapt=glo2nld) & [AliExpress_link2](https://nl.aliexpress.com/item/1005006224025259.html?spm=a2g0o.order_list.order_list_main.4.531a79d2WIOGie&gatewayAdapt=glo2nld) --> the "2pcs incl. remote" sets.
They are detected as TS0601 _TZE200_cpbo62rn.

Software:
The custom Quirk (which I am referring to), is attached to this post.
**The quirk supports:** open, close, stop, set % open/close (aka partially open/close).
**The quirk does not support:** setting open/close limits, luminosity sensor, battery sensor.

_Disclaimer:_ 
1) _This quirk was not written by me, nor did I have any contribution._
2) _I believe its the quick from the feature request post of david-romero._

[TS0601_TZE200_cpbo62rn.txt](https://github.com/user-attachments/files/19113626/TS0601_TZE200_cpbo62rn.txt)

[image]

1liminal1, 2025-03-07:
Thanks for taking the time to write that out. 

The truth is these are just garbage and it was a mistake to buy it on spec. 

If anyone is reading this thinking about buying them... DO NOT

Its not much more for proper switchbots 

The only thing I have kinda got working reliably is the remote, which is almost pointless if you're trying to automate. 

Does anyone know how the remote works? Is it IR or is that just an indicator light on top?

facuzestoa, 2025-05-17:
Thanks for all valuable inputs here, it helped me to have this Chinese one working at least.
Im using quirk provided here by @prairiesnpr  I've just modified following part as command were reversed in my casa.

class TuyaSetUpperLimit(t.enum8):
    """Tuya Set Upper Limit enum."""

    Start = 0x00 # originally this was Stop
    Stop = 0x01

What is in red is not working and green is ok. 

The Control provided by TuyaWindowCoverControl is not working, commands return error regarding cluster 195 and the position is reversed. So I finally disabled the entity in HA
Switch "Inverted" is coming back to ON position after few seconds without any action.

But the most interesting part is working. Motor Direction, Set Upper Limit and State are ok allowing to calibrate from Close to Open, reverse motor direction and controlling it. Good job !

[image]

sp-hy, 2025-07-21:
I'd think twice about using these now.  My one got stuck in the charging cycle.  This burnt out the circuit board and expanded the battery to stop it.  I had to open the unit and disconnect the battery to stop it from overheating.

[image]

[image]

Considering these sit around curtains it's too dangerous. I'll just make my own.