zigpy/zha-device-handlers #2934: [BUG] Xiaomi Dual T2 relay does not show up fields

trapacska, 2024-01-23:
### Bug description

[image]

My intent to use this thing for a garage door. Where Inwould like to have a pulse detached mode enabled with configurable pulse value, and to receive the two reed status from the switch inputs. 

### Steps to reproduce

zhaquirks/xiaomi/aqara/switch_acn047.py

copied this file to custom quirks, restarted HA but I still cant read detached state from S1 and S2 inputs. 

### Expected behavior

It was previously working with z2m in a way that it sent an action single_l1 or single_l2 when thise pins was shorted to COM.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

trapacska, 2024-01-23:
@dmilcah can you help me out please? Ready to provide whatever needed. Dev myself also - still was unable to figure out the quirk world

trapacska, 2024-01-23:
@dmulcahey  can you help me out please? Ready to provide whatever needed. Dev myself also - still was unable to figure out the quirk world

TheJulianJES, 2024-01-23:
Please fill-out the issue template. Remove any custom quirks and provide the device signature and/or device diagnostics.
Both can be seen/downloaded on the device page.

trapacska, 2024-01-24:
@TheJulianJES @dmulcahey updated description with diagnostics, signature and the logs when I paired the device.

trapacska, 2024-01-25:
Updated HA to the latest version and now it seems like that there is a new input type "dropdown", which then showing up fine.

There are two things missing so far: interlock and the events to be received when S1-COM or S2-COM shorted in detached mode.

trapacska, 2024-01-25:
Finally were able to capture the two events for both switches.  Found the issue with the implementation. Will open a PR tomorrow.
[image]
[image]

Until then I was unable to find out why these two lines are not showing up on the UI:
0x02D0: ("interlock", t.Bool, True)
0x00EB: ("pulse_length", t.uint16_t, True)
(here: https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/xiaomi/aqara/switch_acn047.py#L105)

I tried under manage device and set the attribute manually that indeed worked.

@TheJulianJES @dmulcahey can you help me understand what can cause or what can be the reasons for these attributes to not to show up on the UI? I would like to continue investigation but I am stuck with this.

trapacska, 2024-01-26:
OMG, took me 1 day to figure out that it is not possible to add controls on the device screen arbitrarily. These input stuff also hardcoded in[ home-assistant repo](https://github.com/home-assistant/core/blob/7bec5ef6bce997f1ee82a4fc5da0ca02566bdb7f/homeassistant/components/zha/core/cluster_handlers/manufacturerspecific.py#L156)(s). This thing not gonna scale well I guess.

Closing this issue as it provides a partial solution to my original problem but still looking after putting in interlock and pulse_length somewhere else in HA's sources. 

https://github.com/zigpy/zha-device-handlers/pull/2939

closed 2024-01-26 by trapacska

TheJulianJES, 2024-01-27:
The custom entities are added here for now: https://github.com/home-assistant/core/pull/106492/files
The file you linked is for initializing (reading) the attributes on pairing, as ZHA only creates the defined custom entities for the custom attributes if there's a value for the attribute in the attribute cache.