zigpy/zha-device-handlers #2936: [Device Support Request] MOES ZigBee Smart DIY Motorized Roller Blinds

Berenhappinda, 2024-01-24:
### Problem description

I bought a second rolling shades motor.
The first one works perfectly, the second one doesnt.
Functions should be exactly the same.

New device: https://www.aliexpress.com/item/1005005447997612.html?spm=a2g0o.order_list.order_list_main.17.382b1802qFGVs9

Old device (previously ordered and working): https://www.aliexpress.com/item/1005002954586484.html?spm=a2g0o.order_list.order_list_main.49.382b1802qFGVs9

New device (not working): TS0601 by _TZE200_icka1clh
Old device (working):TS0601 by _TZE200_zah67ekd

Signature of working (old)device:
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_icka1clh",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

### Solution description

Can you recreate the quirk for the new device?

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0102",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
[... 307 lines cut ...]
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

Armandogl12, 2024-03-19:
Hello, I have the exact same issue on the same blind motor that arrived today, did you find a solution?? Greetings!

kkggaa45, 2024-03-19:
[image]
Hello, I have the exact same issue on the same blind motor that arrived today, did you find a solution?? Greetings!

Mikboch, 2024-04-02:
I have the same problem as well

Berenhappinda, 2024-04-11:
No solution, i am still waiting for a reply.
A custom quirk needs to be made to resolve the issue.

jhigueras, 2024-04-11:
It's already fixed in home assistant 2024.4.x (which carries zha-quirks 0.0.113). Mine's now working perfectly.

ITMechanic2017, 2024-04-12:
> It's already fixed in home assistant 2024.4.x (which carries zha-quirks 0.0.113). Mine's now working perfectly.

I'm running the latest version of HA core and HA supervisor, support for battery state and cover position still are not working and it doesnt look like this zigby device has yet been added to the support list by zigpy unfortunately

jhigueras, 2024-04-12:
> I'm running the latest version of HA core and HA supervisor, support for battery state and cover position still are not working and it doesnt look like this zigby device has yet been added to the [...]

You're right, i don't see any cover position control. I mean i can use up and down controls and they work perfectly fine :)

closed 2024-10-16 by github-actions[bot]

sammcj, 2025-01-29:
Looks like it wasn't fixed in 2024.4.x, it's still an issue in home assistant 2025