zigpy/zha-device-handlers #2946: [Device Support Request] for Moes Dimmer Switch _TZE204_hlx9tnzb TS0601

RuessAar, 2024-01-28:
### Problem description

This dimmer is not yet part of the "zhaquirks/tuya/ts0601_dimmer.py" file. 

### Solution description

Adding "("_TZE204_hlx9tnzb", "TS0601")," under "MODELS_INFO:" in "class "TuyaSingleSwitchDimmerGP(TuyaDimmerSwitch):" adds the ability to control the on off switch in hassos but the light strength is only controllable in "Manage zigbee device"->"TuyaInWallLevelControlNM (Endpoint id: 1, Id: 0x0008, Type: in)"->"move_to_level (id: 0x0000)"

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_hlx9tnzb",
  "model": "TS0601",
  "class": "TS0601_TZE204_hlx9tnzb.TuyaSingleSwitchDimmerGP"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.1.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
[... 409 lines cut ...]
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaLevelControlManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # input_clusters=[]
            # output_clusters=[33]
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

</details>

RuessAar, 2024-01-31:
I played around some more. Unfortunately with the solution I submitted, only on/off and dimming are possible. Controlling back-light, bulb-type, dimming delay etc. are not possible. I will try to implement these functions into a custom quirk and add it as a comment if it works.

RuessAar, 2024-01-31:
Found the working mqtt equivalent:
```
 {
        fingerprint: tuya.fingerprint('TS0601', ['_TZE200_ip2akl4w', '_TZE200_1agwnems', '_TZE200_la2c2uo9', '_TZE200_579lguh2',
            '_TZE200_vucankjx', '_TZE200_4mh6tyyo', '_TZE204_hlx9tnzb', '_TZE204_n9ctkb6j', '_TZE204_9qhuzgo0']),
        model: 'TS0601_dimmer_1',
        vendor: 'TuYa',
        description: '1 gang smart dimmer',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        configure: tuya.configureMagicPacket,
        exposes: [tuya.exposes.lightBrightnessWithMinMax(), tuya.exposes.countdown(), tuya.exposes.lightType(),
            e.power_on_behavior().withAccess(ea.STATE_SET),
            tuya.exposes.backlightModeOffNormalInverted().withAccess(ea.STATE_SET)],
        meta: {
            tuyaDatapoints: [
                [1, 'state', tuya.valueConverter.onOff, {skip: tuya.skip.stateOnAndBrightnessPresent}],
                [2, 'brightness', tuya.valueConverter.scale0_254to0_1000],
                [3, 'min_brightness', tuya.valueConverter.scale0_254to0_1000],
                [4, 'light_type', tuya.valueConverter.lightType],
                [5, 'max_brightness', tuya.valueConverter.scale0_254to0_1000],
                [6, 'countdown', tuya.valueConverter.countdown],
                [14, 'power_on_behavior', tuya.valueConverter.powerOnBehavior],
                [21, 'backlight_mode', tuya.valueConverter.backlightModeOffNormalInverted],
            ],
        },
        whiteLabel: [
            {vendor: 'Lerlink', model: 'X706U'},
            {vendor: 'Moes', model: 'ZS-EUD_1gang'},
            tuya.whitelabel('Moes', 'ZS-SR-EUD-1', 'Star ring smart dimmer switch 1 gang', ['_TZE204_hlx9tnzb']),
            tuya.whitelabel('Moes', 'MS-105Z', 'Smart Dimmer module', ['_TZE200_la2c2uo9']),
        ],
    },
```

RuessAar, 2024-02-01:
Here are some attributes that zha_toolkit found.
[image]

RuessAar, 2024-02-02:
Here is some information when using no custom quirk. 

ZHA_TOOLKIT:
```
zha_toolkit_version: v1.1.8
zigpy_version: 0.60.7
zigpy_rf_version: 0.37.6
ieee_org:
  - 192
  - 33
  - 193
  - 60
  - 240
  - 56
  - 193
  - 164
ieee: a4:c1:38:f0:3c:c1:21:c0
command: scan_device
command_data: null
start_time: "2024-02-02T06:20:42.675686+00:00"
errors: []
params:
  dir: 0
  tries: 1
  expect_reply: true
  args: []
  read_before_write: true
  read_after_write: true
scan:
  ieee: a4:c1:38:f0:3c:c1:21:c0
  nwk: "0x9fe6"
  model: TS0601
  manufacturer: _TZE204_hlx9tnzb
  manufacturer_id: "0x4417"
  endpoints:
    - id: 1
      device_type: "0x0051"
      profile: "0x0104"
      in_clusters:
        "0x0000":
          cluster_id: "0x0000"
          title: Basic
          name: basic
          attributes:
            "0x0000":
              attribute_id: "0x0000"
              attribute_name: zcl_version
              value_type:
                - "0x20"
                - uint8_t
                - Analog
              access: READ|REPORT
              access_acl: 5
              attribute_value: 3
            "0x0001":
              attribute_id: "0x0001"
              attribute_name: app_version
              value_type:
                - "0x20"
                - uint8_t
                - Analog
              access: READ|REPORT
              access_acl: 5
              attribute_value: 74
            "0x0002":
              attribute_id: "0x0002"
              attribute_name: stack_version
              value_type:
                - "0x20"
                - uint8_t
[... 374 lines cut ...]
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_hlx9tnzb",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

RuessAar, 2024-02-02:
Also here the zha diagnostic when using no quirk:

```
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.1.6",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "6.1.63-haos-raspi",
    "supervisor": "2023.12.1",
    "host_os": "Home Assistant OS 11.4",
    "docker_version": "24.0.7",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "adaptive_lighting": {
      "version": "1.20.0",
      "requirements": [
        "ulid-transform"
      ]
    },
    "better_thermostat": {
      "version": "1.4.0",
      "requirements": []
    },
    "duckdns_ipv4_ipv6": {
      "version": "1.1.1",
      "requirements": []
    },
    "hacs": {
      "version": "1.34.0",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "zha_toolkit": {
      "version": "v1.1.8",
      "requirements": [
        "pytz"
      ]
    },
    "powercalc": {
      "version": "v1.10.0",
      "requirements": [
        "numpy>=1.21.1"
      ]
    },
    "thermal_comfort": {
      "version": "2.2.2",
      "requirements": []
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "after_dependencies": [
      "onboarding",
      "usb"
    ],
    "codeowners": [
      "@dmulcahey",
      "@adminiuga",
[... 358 lines cut ...]
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      },
      "242": {
        "device_type": {
          "name": "PROXY_BASIC",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

EzEsAz, 2024-03-16:
HI [RuessAar]! I have similar dimmer. Did you find a solution for Controlling back-light, bulb-type, dimming delay etc. ?

jodasm, 2024-09-09:
I also have this dimmer.  I could not get "start_up_on_off (id: 0x4003)" to work on the "TuyaOnOff (Endpoint id: 1, Id: 0x0006, Type: in)" cluster.

highseen, 2024-12-05:
I am using the same model. Could pair it easily only to see that the dimmer doesn't fully comply to zigbee standards.
Would very much appreciate if the first increment on/off and dimming would get integrated!
Need any help?

kaiontour, 2025-01-11:
hello everybody, for me on/off and dimm is working with the custom quirk. 

I have another problem. I user E27 bulbs and when switch on or I want to dimm the bullbs make a load noise. After a while they are damaged. Is somebody using bulbs? And if so, which one are working?

bergerx, 2025-01-19:
@kaiontour do you mind sharing the custom quirk you have?

Dan1jel, 2025-03-30:
> @kaiontour do you mind sharing the custom quirk you have? (ah please ignore as ts already in the issue decription)

If you have same version as i did, i modified this and added my device model in the quirk to make it work, search for ```_TZE204_hlx9tnzb``` to see where. 

```
"""Tuya based touch switch."""

from zigpy.profiles import zgp, zha
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import NoManufacturerCluster, TuyaDimmerSwitch
from zhaquirks.tuya.mcu import (
    TuyaInWallLevelControl,
    TuyaLevelControlManufCluster,
    TuyaOnOff,
    TuyaOnOffNM,
)

class TuyaInWallLevelControlNM(NoManufacturerCluster, TuyaInWallLevelControl):
    """Tuya Level cluster for inwall dimmable device with NoManufacturerID."""

# --- DEVICE SUMMARY ---
# TuyaSingleSwitchDimmer: 0x00, 0x04, 0x05, 0xEF00; 0x000A, 0x0019
# TuyaDoubleSwitchDimmer: 0x00, 0x04, 0x05, 0xEF00; 0x000A, 0x0019
# - Dimmer with Green Power Proxy: Endpoint=242 profile=41440 device_type=0x0061, output_clusters: 0x0021 -
# TuyaSingleSwitchDimmerGP: 0x00, 0x04, 0x05, 0xEF00; 0x000A, 0x0019
# TuyaDoubleSwitchDimmerGP: 0x00, 0x04, 0x05, 0xEF00; 0x000A, 0x0019
# TuyaTripleSwitchDimmerGP: 0x00, 0x04, 0x05, 0xEF00; 0x000A, 0x0019

class TuyaSingleSwitchDimmer(TuyaDimmerSwitch):
    """Tuya touch switch device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_dfxkcots", "TS0601"),
            ("_TZE200_whpb9yts", "TS0601"),
            ("_TZE200_ebwgzdqq", "TS0601"),
            ("_TZE200_9i9dt8is", "TS0601"),
            ("_TZE200_swaamsoy", "TS0601"),
            ("_TZE200_0nauxa0p", "TS0601"),
            ("_TZE200_la2c2uo9", "TS0601"),
            ("_TZE200_1agwnems", "TS0601"),  # TODO: validation pending?
            ("_TZE200_9cxuhakf", "TS0601"),  # Added for Mercator IKUU SSWM-DIMZ Device
            ("_TZE200_a0syesf5", "TS0601"),  # Added for Mercator IKUU SSWRM-ZB
            ("_TZE200_p0gzbqct", "TS0601"),
            ("_TZE200_w4cryh2i", "TS0601"),
            ("_TZE204_dcnsggvz", "TS0601"),
        ],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=0x0051
            # device_version=1
            # input_clusters=[0, 4, 5, 61184]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaLevelControlManufCluster.cluster_id,
                ],
[... 263 lines cut ...]
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                    TuyaInWallLevelControlNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                    TuyaInWallLevelControlNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

rudizl, 2025-06-01:
Any chances to add official support for that devices?