zigpy/zha-device-handlers #2949: [BUG] SONOFF ZBMINIL2 potentially requiring quirk

Migsi, 2024-01-29:
### Bug description

I've recently got my hands onto an SONOFF ZBMINIL2, which is generally working fine, BUT it appears to be improperly recognized as battery powered (or unknown). Also I've noticed it is impossible to add it to a Zigbee group (it doesn't get listed at the group setup page).
Being unaware of the inner workings of ZHA, I see potential these two issues could be linked.

### Steps to reproduce

1. Link the device
2. Check the available entities (or the signature)
3. Try to setup a Zigbee group with this device

### Expected behavior

I'd like the device to not have a (useless) battery indicator and be able to setup a Zigbee group containing it.

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4742, maximum_buffer_size=82, maximum_incoming_transfer_size=1024, server_mask=11264, maximum_outgoing_transfer_size=1024, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0002",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0006",
        "0x0007",
        "0x0b05",
        "0xfc57"
      ],
      "output_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "SONOFF",
  "model": "ZBMINIL2",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Unknown",
    "version": "2024.1.5",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": false,
    "arch": "x86_64",
    "timezone": "Europe/Vienna",
    "os_name": "Linux",
    "os_version": "6.1.63-production+truenas",
    "run_as_root": true
  },
  "custom_components": {
    "thermal_comfort": {
[... 349 lines cut ...]
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
None
```

</details>

### Additional information

If these two issues are unrelated, I apologize for my wild guess and will happily reduce this ticket to the power source problematic only.
If someone could provide me a hint on how to create a quirk to overwrite given property, I'd be happy to submit a PR.

noseshimself, 2024-06-15:
It's not only that the power source is "battery or unknown"; the devices are potential routers but without being "mains" powered they are not eligible. I had to put another device in the area as Zigbee repeater which the switches were intended to be.

Migsi, 2024-06-30:
> It's not only that the power source is "battery or unknown"; the devices are potential routers but without being "mains" powered they are not eligible. I had to put another device in the area as Zigbee repeater which the switches were intended to be.

AFAIR the device wrongly advertises itself as non mains powered, even though it is. It is not usable as router though, due to weird limitations Sonoff made on this model. Having ZHA applying a quirk recognizing the power source as mains should eliminate the misleading/invalid battery indicator, as long as Sonoff doesn't provide a fix.
I texted their support in hopes to get that rolling, but had no luck yet. The report was "handed over to the programming team" (or something along the lines), and never to be heard of again. :(

closed 2025-01-03 by github-actions[bot]