zigpy/zha-device-handlers #2950: [Device Support Request] mmWave presence sensor TS0601 by _TZE204_muvkrjr5

Goaheadz, 2024-01-29:
### Problem description

So I just received a new mmwave sensor TS0601 by _TZE204_muvkrjr5 from Aliexpress.
I need a quirk file to make parameters available in Home Assistant..no experience to do this :(

Can someone help?

[image]

https://pt.aliexpress.com/item/1005006412041826.html?spm=a2g0o.order_list.order_list_main.29.52afcaa4H5VTTL&gatewayAdapt=glo2bra

### Solution description

help needed.

INFO (MainThread) [zigpy.endpoint] [0xb905:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[4, 5, 61184, 0], output_clusters=[25, 10])

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_muvkrjr5",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
[... 370 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

Goaheadz, 2024-01-30:
Device signature is the same as _TZE204_qasjif9e just like Simpledescriptor

sloaiza86, 2024-04-23:
> I am using for now:
> 
> [ts0601_motion.txt](https://github.com/zigpy/zha-device-handlers/files/14100362/ts0601_motion.txt)
> 
> Copy paste text to the .py file, inside Home assistant
> 
> Dont know what GPP means, so I add it to "variant 1" and "variant 1 gpp" Can someone explain?
> 
> Restart HA and got the entities. ![image](https://private-user-images.githubusercontent.com/13796554/300860041-09a0792f-60c7-4a65-ab4a-81014d50f671.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM5MDEyNzEsIm5iZiI6MTcxMzkwMDk3MSwicGF0aCI6Ii8xMzc5NjU1NC8zMDA4NjAwNDEtMDlhMDc5MmYtNjBjNy00YTY1LWFiNGEtODEwMTRkNTBmNjcxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDIzVDE5MzYxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNiM2FiZGVhYmE3MGU5MGM0YTNkMDE0ZWZhMDI3OWIwYzk5NDRhOWM4Njg1ZTA4ZGQwOWU3NDczOTM5ODg3NmQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.pr2z34N6zzJ07Qf1xNcngdsyYiWxb6ejVVWqphT588E) I am not at home (doing this remotely) so I dont know if entities adjustment works. What I can see is that sensor is constantly changing from clear to occupancy. Sensor is currently near the floor, so I have to check that later. I will also check parameters from sensor using Bluetooth and hi-link radar tool in case the constant change of status is or is not a integration problem.

Have you had any further advance in making this device compatible, or is this your last version?

closed 2024-10-27 by github-actions[bot]