zigpy/zha-device-handlers #2951: [Device Support Request] smart thermostat TS0601

dekar91, 2024-01-30:
### Problem description

The TS0601 is not supported now. When connected, it shows no entities.
The current workaround is to use the device through Tuya integration(and Tuya gateway).
I would prefer to connect it directly to HA.

### Solution description

The thermostat recognises and works directly with HA directly.

### Screenshots/Video

[image]

### Device signature

<details><summary>Device signature</summary>

```json
"node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_6rdj8dzm",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details><summary>Debug log</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.1.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "6.1.63-haos-raspi",
    "supervisor": "2023.12.1",
    "host_os": "Home Assistant OS 11.4",
    "docker_version": "24.0.7",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.33.0",
      "requirements": [
[... 296 lines cut ...]
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

jacekk015, 2024-03-13:
https://github.com/jacekk015/zha_quirks
ts0601_trv_me167.py

dekar91, 2024-03-13:
> https://github.com/jacekk015/zha_quirks ts0601_trv_me167.py

Thanks, @jacekk015! This quirk works perfectly!