zigpy/zha-device-handlers #2956: [Device Support Request] ZBMINIL2 change to momentary switch

ebarriosjr, 2024-01-30:
### Problem description

I set up a Sonoff ZBMINIL2 as a momentary switch in my wall. The device is configured to act as a momentary switch (triple click the button on the device) and it works with the wall momentary switch, but it appears as a switch in home assistant and I can't find any way to modify the configuration to treat it as a momentary switch,

### Solution description

I would like to be able to change the configuration to work as a momentary switch.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4742, maximum_buffer_size=82, maximum_incoming_transfer_size=1024, server_mask=11264, maximum_outgoing_transfer_size=1024, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0002",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0006",
        "0x0007",
        "0x0b05",
        "0xfc57"
      ],
      "output_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "SONOFF",
  "model": "ZBMINIL2",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.1.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "6.1.71-haos",
    "supervisor": "2023.12.1",
    "host_os": "Home Assistant OS 11.4",
    "docker_version": "24.0.7",
[... 441 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

closed 2024-08-04 by github-actions[bot]

uschindler, 2025-07-22:
I think for this issue the additional config option from the SONOFF cluster needs to be added as a quirk. I will check if the ZBMINIR2 Quirk can also be applied here (adding the missing option to chnage the "trigger mode".

uschindler, 2025-07-22:
reopen