zigpy/zha-device-handlers #2959: [Device Support Request] Moes ZM-105-M - TS0601 von _TZE204_bxoo2swd Tuya

gerdy001, 2024-01-31:
### Problem description

Hello,
When trying to add these zigbee dimmer modules via ZHA it does not add any entities, only the device.

### Solution description

Are you able to assist with support for this device/entities?

Thanks in advance,
Gerhard

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

{
"node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_bxoo2swd",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

### Diagnostic information

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.1.6",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "6.1.71-haos",
    "supervisor": "2023.12.1",
    "host_os": "Home Assistant OS 11.4",
    "docker_version": "24.0.7",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.34.0",
      "requirements": [
[... 329 lines cut ...]
        }
      }
    }
  }
}

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

gerdy001, 2024-02-01:
https://github.com/zigpy/zha-device-handlers/pull/2903
If I Understand, it is already supported. Why my Dimmer is not showing any entities in HomeAssistant?

javicalle, 2024-02-04:
New version not released yet.
It will be included in HA 2024.2

AlexMKX, 2024-02-13:
At this moment it works great. Device itself is relatively good comparing to many other dimmers - always good response time.
However, it stops working after some time (like 12-24 hours), which results in : No timeouts, commands are executed, brightness and on/off is set successfully. However, it does not change anything. Which means : I power on dimmer, Z2M says dimmer powered on, but, actually it is not.
The reconfig fixes this behavior untill next 12-24 hours.

DefinitelyADev, 2024-06-20:
> At this moment it works great. Device itself is relatively good comparing to many other dimmers - always good response time.
>
> However, it stops working after some time (like 12-24 hours), which [...]

Does this still keep happening? I got one to test it and see how it goes. 

Has anyone figured out a way to make it cycle dimming? Because now you have to press and hold the button to start dimming up, then press and hold again, to dim it down. I was wandering if there is a value to configure it.

Also, I think this device has more features in the moes app, like timers and stuff. Would it be possible to make it compatible? (I am a complete noob with zigbee internals and integrations).

Cheaternl, 2024-10-16:
Unfortunately I have the same issue, where the dimmer becomes unresponsive after a couple of hours (I didn't measure it). 
Within that state, the dimmer completely doesn't act on what you tell it to do. I bought 2 of them from the same seller, and both behave the same.

Cheaternl, 2024-12-24:
I can retest, after the holidays have passed.  I assume I will get into trouble if I let the diner-table lamp in an uncontrollable state during these days (if the issue still persists) 😅

A4cp, 2024-12-24:
Merry Christmas mate and take your time! 🎄🎅🎁

Cheaternl, 2025-07-20:
I retested both my Moes 2 Gang Dimmer Modules.
One of it, completely failed to react on any changes in state I performed.
The other one only worked on 1 of the actors to switch on or off. 

I lost faith in this dimmer and I'm looking into other brand dimmers.