zigpy/zha-device-handlers #2964: [Device Support Request] Develco/Frient intelligent keypad KEPZB-110

eddiekwak202, 2024-02-04:
### Problem description

What I am able to do:
- Arm/disarm in differnt modes from the keypad itself and through Home Assistant.
- Reading the code typed into the keypad, and the (dis)arming mode via the zha_event.

What will not work/is not available:
- entry/exit delay, although I can send it manually through the "manage device" pop-up.
- tamper detection
- panic button (right now it just triggers the alarm)

### Solution description

Being able to use the keypad as it is intendet to use.
This would include:
- entry/exit delay
- tamper detection
- working battery readout (or only a "low battery" funtion)
- (a better way to read the code that is send to ZHA)

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4117, maximum_buffer_size=82, maximum_incoming_transfer_size=1500, server_mask=11264, maximum_outgoing_transfer_size=1500, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0xc0c9",
      "device_type": "0x0001",
      "input_clusters": [
        "0x0005",
        "0x0006"
      ],
      "output_clusters": []
    },
    "44": {
      "profile_id": "0x0104",
      "device_type": "0x0401",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0500",
        "0x0b05"
      ],
      "output_clusters": [
        "0x0003",
        "0x000a",
        "0x0019",
        "0x0501"
      ]
    }
  },
  "manufacturer": "frient A/S",
  "model": "KEPZB-110",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

Link to the Develco technical documentation, including the used Zigbee clusters:

https://www.develcoproducts.com/media/2032/kepzb-110-technical-manual-keypad.pdf

masi, 2024-03-08:
Entering a code follow by a mode button, does not trigger a zha_event for my me.

eddiekwak202, 2024-03-09:
@masi Have you reconnecting the device? For me it also didn't work the first time I connected the keypad. After a quick reconnect it worked for me.

masi, 2024-03-09:
> @masi Have you reconnecting the device? For me it also didn't work the first time I connected the keypad. After a quick reconnect it worked for me.

No. Will try to see if it makes any difference.

Hopefully as seen in https://github.com/zigpy/zha-device-handlers/issues/901 the device some others get events.

masi, 2024-03-09:
@eddiekwak202 issue https://github.com/home-assistant/core/issues/72828 is bad news for tamper support.

If I got it right ZHA will not no in advance if a device has tamper support. So the decision was not to add it because for mere motion dectors (used for lighting and not security) will not have a tamper detection. The reasoning was that an entity for an unsupported feature would trigger more support tickets.

Perhaps quirks v2 will get something like @puddly envisioned here:
https://github.com/home-assistant/core/issues/72828#issuecomment-1172462271

eddiekwak202, 2024-03-16:
@masi well that is too bad. But the tamper function should always be a last line of defense anyway (except for outdoor cameras). But did you manage to get your keypad working already?

masi, 2024-03-16:
@eddiekwak202 Yes. After a factory reset the keypad fires events after pressing one of the mode buttons.

To be honest I have no clue what the manual tells me about EZ modes initiator and target. And despite the manuals says the keypad will scan only channels 11-24 it has joined my network on channel 25.

What I did not manage to do was to change the entry/exit delay setting. I couldn't find any attribute in one of the clusters that looked like the delay settings.

eddiekwak202, 2024-03-16:
@masi yeah the entry/exit only works while calling it manually for me. Then it is under mange zigbee device/IasAce/change_panel_status. But I dont know if it is possible to call this in a service call in home assistant.

masi, 2024-03-16:
@eddiekwak202 I think you can send this command with https://github.com/mdeweerd/zha-toolkit

Been reading the Zigbee Cluster Library rev8 to understand how keypads are supposed to work. As far as i understood you can get some parts of the Zigbee Intruder Alarms System (IAS) working, but it will be tough. Zigpy and/or ZHA simply do not implement enough of the spec.

ZHA should act as Control and Indicating Equipment (CIE), but it does not:
https://github.com/zigpy/zigpy/issues/419

eddiekwak202, 2024-03-26:
@masi well I jut read the article...but it seems that there is just not so much interest in the alarm part of zha. Although there are some parts implemented, and I think it can be really usefull and can exceed beyond the limits of a regular system.

sbusceti, 2024-08-02:
Hi, is there a way to configure the keypad to emit the exit/entry delay sounds without having to manually send the command each time?

masi, 2024-08-31:
> Hi, is there a way to configure the keypad to emit the exit/entry delay sounds without having to manually send the command each time?

I don't think that is the way Zigbee IAS keypads are supposed to work. IMHO it is up to the "security central" to coordinate the devices.

What is missing is a ZHA part that implements the "central".

Perhaps its is possible the get the delay from an Alarmo event. At least for MQTT it sends out the delays to the queue.

sbusceti, 2024-10-15:
Would it be possible, through a custom quirk, to read the bit related to tamper(or any other attribute) and expose this information as a binary sensor on Home Assistant?

masi, 2024-10-18:
Possibly.

AFAIK quirks v2 are meant to make this easy or easier than now. But I haven't heard about the new way to write a custom quirks for a while.

Not sure if I would be able to write a quirk though. I'm not a Python dev.

closed 2025-08-06 by puddly