zigpy/zha-device-handlers #2968: [Device Support Request] TS0601 _TZE200_g9a3awaj

jezzaaa, 2024-02-08:
### Problem description

I have a Tuya-based ZigBee thermostat purchased from Aliexpress. The device TS0601 _TZE200_g9a3awaj has no support in Quirks. The device packaging lists the manufacturer as Ecobuy smart controls Co Ltd, and the model number is T5Z.

However, I was able to get it to partially work by hacking the device name into a local copy of ts0601_electric_heating.py. The current temp and set temp are showing in HA, although the set temp is scaled by a factor of 10 (eg 215C instead of 21.5C). The Idle/Heating status is reversed (shows Heating when the thermostat turns off and removes the flame symbol). I've tweaked the code to get the set temp displaying correctly, and I could probably work out how to reverse the sense of the Idle/Heating setting. But other issues would remain.

Probably the biggest problem is that setting anything in HA doesn't reflect on the thermostat. If I change the temperature on HA, the change isn't reflected on the thermostat. If I change the mode from Heat to Off, the thermostat doesn't turn off.

Aliexpress link [here](https://www.aliexpress.com/item/1005006241271065.html). The device is battery powered, and has dry Normally Open (NO) contacts, rated 24-230VAC up to 5A.

### Solution description

A new quirk is required for this device. I can code, but I'm not familiar with how HA's codebase hangs together, nor do I know anything about how to interact with ZigBee devices. I'd gladly work on this if I had a few pointers on how to implement the missing features.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0301",
          "input_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0x0201",
            "0x0204",
            "0xef00"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        }
      },
      "manufacturer": "_TZE200_g9a3awaj",
      "model": "TS0601"
    }
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.2.0",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.12.1",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Australia/Sydney",
[... 669 lines cut ...]
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.THERMOSTAT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    JezManufCluster,
                    JezThermostat,
                    JezUserInterface,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
```

</details>

jezzaaa, 2024-02-11:
This appears to be a re-branded [Avatto model ZWT07](https://avatto-smart.com/product/zigbee-smart-thermostat-zwt07/). There's [support in Z2M](https://www.zigbee2mqtt.io/devices/ZWT07.html) for this model.

douglascrc-git, 2024-02-16:
> This appears to be a re-branded [Avatto model ZWT07](https://avatto-smart.com/product/zigbee-smart-thermostat-zwt07/). There's [support in Z2M](https://www.zigbee2mqtt.io/devices/ZWT07.html) for this model.

I have the same device.

After testing the quirk you provided, I noticed that the device appeared in Home Assistant (HA). However, I encountered a limitation – while I could read the values, I couldn't adjust the temperature or control the device's power state.

Additionally, I observed an anomaly: when the set point temperature is lower than the current temperature, the device begins heating, but when the set point is higher, the device remains idle. This suggests there may be an issue with the code.

I will now attempt to troubleshoot and rectify the problem.

jezzaaa, 2024-02-17:
> After testing the quirk you provided,

I didn't provide any quirk here. I only requested support for an existing quirk to support this device.

> I noticed that the device appeared in Home Assistant (HA). However, I encountered a limitation – while I could read the values, I couldn't adjust the temperature or control the device's power state.

Yes, that's what I found. The values seemed to be read-only.

> Additionally, I observed an anomaly: when the set point temperature is lower than the current temperature, the device begins heating, but when the set point is higher, the device remains idle. This suggests there may be an issue with the code.

I think what you're saying here is that the heating/idle are reversed? Is it just the status shown in HA that's reversed, or is it the thermostat itself actually turning on the heating when its set temp is lower? If it's the latter then this would likely happen even without connecting to HA, and would indicate a fault in the thermostat.

I took a look at a fork of the ZHA quirks by @jacekk015; one that included an Avatto thermostat quirk. It didn't include our model of thermostat, but it's is similar enough to one of the supported models (_TZE200_2ekuz3dz) that I managed to get it to work correctly for me. One of the issues I had to address was that the heating/idle status was being reversed in the existing code so I had to remove that reversal for my model to have the correct status displayed. Most importantly, the Avatto quirk allowed setting, not just reading, the power state and the set temp.

Perhaps you could give the Avatto quirk a try, with my mods that I've submitted to @jacekk015 for inclusion: https://github.com/jacekk015/zha_quirks/issues/49

jezzaaa, 2024-02-18:
The "2ekuz3dz" model on which I based my mods is supported in the mainline code, in the file ts0601_electric_heating.py. I had a go at doing the same mods to this file, but I was unsuccessful - the result was read-only, so I couldn't change the set temp. There are significant differences between how "2ekuz3dz" is supported in the "ts0601_electric_heating.py" file and jacekk015's "ts0601_thermostat_avatto.py" file, so much so that I wonder if the "2ekuz3dz" thermostat actually works with the mainline code.

Jacekk015's quirk has a lot of devices in the one file, which made his pull request unable to be tested and accepted for inclusion in mainline. I'd really love to get support for this thermostat added, but without help from knowledgeable folks, I don't know how I can make progress. All I can think of doing is to make a copy of my hacked working version, stripping out support for the handful of other thermostats, and trying to get it accepted on its own. I don't know that I have the skills to do this.

douglascrc-git, 2024-02-18:
Thank you for you quick answer! 

> I think what you're saying here is that the heating/idle are reversed? Is it just the status shown in HA that's reversed, or is it the thermostat itself actually turning on the heating when its set [...]

I was talking about the status in HA being reversed, not the thermostat itself. 

> I took a look at a fork of the ZHA quirks by @jacekk015; one that included an Avatto thermostat quirk. It didn't include our model of thermostat, but it's is similar enough to one of the supported models (_TZE200_2ekuz3dz) that I managed to get it to work correctly for me. One of the issues I had to address was that the heating/idle status was being reversed in the existing code so I had to remove that reversal for my model to have the correct status displayed. Most importantly, the Avatto quirk allowed setting, not just reading, the power state and the set temp.
> 
> Perhaps you could give the Avatto quirk a try, with my mods that I've submitted to @jacekk015 for inclusion: [jacekk015/zha_quirks#49](https://github.com/jacekk015/zha_quirks/issues/49)
>

I gave a try to the file you [uploaded](https://github.com/jacekk015/zha_quirks/files/14230994/ts0601_thermostat_avatto.zip), sadly it did not work for me. The issue with the status being reversed was solved, nevertheless it is still imposible to change from HA the target temperature, turn  on/off , etc. 

I have exactly the same device you have.  What version of HA OS do you have?

This is how the device is shown in HA for me:
[image]

jezzaaa, 2024-02-18:
I'm running HAOS 11.5 with core 2024.2.2. Here's my device info page:

[image]

I'm curious that your device isn't showing max and min heat settings under "Configuration" like mine is. Instead you have "Compensacion" (compensation?) whereas mine does not. (This is different from "Number Temperature Calibration" which we both have under "Controls"). I'm wondering if you might have another quirk being loaded? I don't think my quirk has compensation.

After adding your quirk did you remove and re-add your device? I don't know if this is strictly necessary, but I've seen it recommended in some forum posts. When I add/change a quirk, I just restart Home Assistant and that seems to be sufficient most of the time.

douglascrc-git, 2024-02-19:
> I'm running HAOS 11.5 with core 2024.2.2. Here's my device info page:
> 

I had a older version, maybe it is the reason my HA instalation displays different configuration.

> I'm curious that your device isn't showing max and min heat settings under "Configuration" like mine is. Instead you have "Compensacion" (compensation?) whereas mine does not. (This is different [...]

I believed that too, sadly, after seeking, I could not find another quirk assigned to the thermostat. So, that was not the cause.

> After adding your quirk did you remove and re-add your device? I don't know if this is strictly necessary, but I've seen it recommended in some forum posts. When I add/change a quirk, I just restart Home Assistant and that seems to be sufficient most of the time.

 I deleted the device and tried to add it again, but HA was not able to link the device.  So, I try in another raspberry with an older version of HA and the quirk work perfectly! Nevertheless, HA does not show the same configuration that you have. 

I appreciate your help!

[image]

Next, I will try it again in a newer version of HA, because I'd like to have the same parameters you have.

douglascrc-git, 2024-03-08:
> I'm running HAOS 11.5 with core 2024.2.2. Here's my device info page:

I finally updated my HAOS 
[image]

But the device does not work! The HVAC is unkown 

[image]

Can you please share the quirk you have installed?

jezzaaa, 2024-03-09:
> Can you please share the quirk you have installed?

It hasn't changed. It's the same one I put in my issue in Jacekk015's repo: https://github.com/jacekk015/zha_quirks/issues/49

darknessblade, 2024-05-11:
> Code has been merged. Quirk updated https://github.com/jacekk015/zha_quirks

Which one of the custom quirks do I need to use? I have gotten one as well, and it shows up as a temp-sensor instead of a thermostat

jacekk015, 2024-05-11:
@darknessblade
`
2024-05-11 14:14:58.300 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: {1, 242} {1}
`
Is that a new device? Signature is OK, but it complains about an extra 242 endpoint... which those device normally have.

darknessblade, 2024-05-11:
> @darknessblade `2024-05-11 14:14:58.300 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: {1, 242} {1}` Is that a new device? Signature is OK, but it complains about an extra 242 endpoint... which those device normally have.

This is the one I bought roughly a week ago:
aliexpress.com/item/1005006241271065.html?

[After discounts it was like 9 euro]

was Planning to use it as something of a remote for my IR panel, which is connected to a different heating panel. 
------
Production date on the package says: "May 4 2024"
It might also be the device that is defective [software defect]. because the smartelife app also has issues with it.

jacekk015, 2024-05-11:
Probably Chinese friends changed the endpoints and left same signature.
You could check updated ts0601_thermostat_avatto.py file
https://github.com/jacekk015/zha_quirks
Should be OK.

MattWestb, 2024-05-11:
The MCU is the same and its having the the "tuya device ID" is coming from the MCU and the Zigbee module is reading it the its booting up.
So they have changing the Zigbee module or the firmware on it but the MCU is the same.

darknessblade, 2024-05-11:
> Probably Chinese friends changed the endpoints and left same signature. You could check updated ts0601_thermostat_avatto.py file https://github.com/jacekk015/zha_quirks Should be OK.

A lot more entities shows up now only the thermostat entity is not visible [Hvac on/off works]

[home-assistant_2024-05-11T20-09-23.147Z.log](https://github.com/zigpy/zha-device-handlers/files/15284174/home-assistant_2024-05-11T20-09-23.147Z.log)

darknessblade, 2024-05-11:
> The MCU is the same and its having the the "tuya device ID" is coming from the MCU and the Zigbee module is reading it the its booting up. So they have changing the Zigbee module or the firmware on it but the MCU is the same.

Might be the case, as smartelife doesn't like it, with it not properly detecting the thermostat.

Which is the benefit of HA, that it is more versatile and can work with nearly ANY entity/quirk

closed 2024-11-14 by github-actions[bot]

jdbrookes, 2025-01-16:
I've purchased this model of thermostat to test as a controller for a gas boiler heating system - it's not detected in zha currently (no entities for the device) but after configuring zha for the quirk in this repository the expected entities appear. Everything appears to be working as expected but will be testing this for a while before it goes into "production". Many thanks @jacekk015

vegos, 2025-01-28:
After your information here I managed to pair the thermostat.

[image]

but as you can see, I get "unknown" on thermostat.
Lock and Power works fine.

[image]

Any info?
Thanks!

darknessblade, 2025-01-28:
> After your information here I managed to pair the thermostat.
> 
> [image]
> 
> but as you can see, I get "unknown" on thermostat. Lock and Power works fine.
> 
> [image]
> 
> Any info? Thanks!

Try to restart HASS, and check the thermostat again. not sure what causes it, usually a restart fixes thermostat isues

vegos, 2025-01-28:
[image]

No luck after a lot of reboots and reinstall of the thermostat.
Strange...

darknessblade, 2025-01-29:
> [image]
> 
> No luck after a lot of reboots and reinstall of the thermostat. Strange...
> 
> [image]

Which version do you have? [Electric Gas or water]

vegos, 2025-01-29:
"Tuya Zigbee3.0 Smart Thermostat for Water Gas Boiler Valve Smart WiFi Wireless Temperature Controller Voice with Alexa Google Home"

https://banggood.onelink.me/zMT7/8defharl

darknessblade, 2025-01-29:
> "Tuya Zigbee3.0 Smart Thermostat for Water Gas Boiler Valve Smart WiFi Wireless Temperature Controller Voice with Alexa Google Home"
> 
> https://banggood.onelink.me/zMT7/8defharl

That is a different model than mine.
Its also ON/OFF **/** GAS, and not electric heating.

I have 2 of those myself [generic brand] and they work fine without issues, with no need for custom quirks.
If you are using custom quirks try removing them. 
Also HARD-reboot Home-assistant.

jdbrookes, 2025-01-29:
> "Tuya Zigbee3.0 Smart Thermostat for Water Gas Boiler Valve Smart WiFi Wireless Temperature Controller Voice with Alexa Google Home"
> 
> https://banggood.onelink.me/zMT7/8defharl

That seems to be the same model as mine (I ordered from aliexpress), it's detected as such in ZHA:

  Device info
  TS0601
  by _TZE200_g9a3awaj
  Connected via [dresden elektronik Conbee II]
  Zigbee info
  IEEE: a4:c1:38:etcetcetc
  Nwk: 0x220b
  Device Type: EndDevice
  LQI: Unknown
  RSSI: Unknown
  Last seen: 2025-01-29T08:51:43
  Power source: Battery or Unknown
  Quirk: ts0601_thermostat_avatto.Beok2

What do you see in your own Zigbee Info section? just make sure it's the right quirk and that you downloaded the quirk correctly from github - you should see in your Home Assistant log after reboot that the quirk was loaded correctly.

Other than that, have you checked if your zigbee coordinator / stick has the latest firmware update? I assume you have other zigbee devices connecting ok?

vegos, 2025-01-29:
[image]

darknessblade, 2025-01-29:
Same model:
Quirk: ts0601_thermostat_avatto.Beok2

Avatto: [still uses a custom quirk as far as I can see]
Quirk: ts0601_thermostat_avatto2.Avatto2

vegos, 2025-01-29:
I tried with avatto2 quirk but it's not working. As I can see in the py file there is no definition for _TZE200_g9a3awaj.

Where i can download yours?

darknessblade, 2025-01-29:
> I tried with avatto2 quirk but it's not working. As I can see in the py file there is no definition for _TZE200_g9a3awaj.
> 
> Where i can download yours?

I am not sure if the Avatto one works with yours, as its specifically for a ELECTRIC heating panel.

here is a link to my request I made: 
https://github.com/zigpy/zha-device-handlers/issues/2589
you should be able to get the custom quirks file there.

vegos, 2025-01-29:
No luck. Tried everything...

darknessblade, 2025-01-30:
> No luck. Tried everything...

If you have a CHEAP tuya zigbee hub you could use that one for troubleshooting. {I have one myself and only use it to troubleshoot devices that have issues}

example:
Moes thermostat gives "set temp= 100000000000C" for 1ms,
Same issue happens in the tuya app as well
result: return it for a full refund

vegos, 2025-01-30:
I have a Sonoff ZBdongle-E Smart usb dongle that is spare and I can play with it. 
Don't know where to start lol

darknessblade, 2025-01-30:
> I have a Sonoff ZBdongle-E Smart usb dongle that is spare and I can play with it. Don't know where to start lol

That is not what I meant. you need something like this:
https://aliexpress.com/item/1005005703607711.html?

I have one [from the Bundle deals, costed like 5 euro's]
and only use it for troubleshooting.

vegos, 2025-01-30:
I find some strange behaviour.

[image]

Reading that, I get "None" as reply.
This is the status of thermostat (powered on or off).

Writing 0 or 1 can turn it on or off. But I can't read the status.

I tried that with other attributes as well and I got the same behaviour.

Is it normal? I think that it must be the same attribute for read and write for a specific purpose....

mforrester1975, 2025-04-17:
FYI I used the updated quirk and had no thermostat after reconfigure and rebooting HA inc hard reboot.

I then removed a battery from the Zigbee unit and waited 10 seconds, added back in and let the device reboot, thermostat instantly appeared and could be seen by HA