zigpy/zha-device-handlers #2970: [Device Support Request] TS0601 _TZE204_ue3rzddr

jollyekb, 2024-02-09:
### Problem description

device not supported in HA

### Solution description

device is supported

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_ue3rzddr",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.1.6",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "aarch64",
[... 451 lines cut ...]
</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

https://www.alibaba.com/product-detail/Smart-Knob-Dimmer-Switch-WiFi-Tuya_1600574868597.html

javicalle, 2024-02-10:
I am not sure if can be fully suported but I would suggest to try to add to the `TuyaSingleSwitchDimmerGP` quirk:
https://github.com/zigpy/zha-device-handlers/blob/ab0d248defda933019c98ddab451b5969fe99c16/zhaquirks/tuya/ts0601_dimmer.py#L145-L154

Once added, if not working, the logs can give us some extra information

jollyekb, 2024-02-11:
where i can find this file on my HA? or i have to rebuild HA?

javicalle, 2024-02-11:
If you don't know how, this could be a good starting point:
https://community.home-assistant.io/t/zigbee-guide-how-to-setup-local-custom-device-handler-quirks-in-zha-integration/683473

You would need to create a local copy of `zha-device-handlers/zhaquirks/tuya/ts0601_dimmer.py ` file and then add your device to the `MODELS_INFO` list.

Save changes, restart HA and check if the device is loading the quirk (you can check it in the device view)

jollyekb, 2024-02-11:
Its working now, but i see 2 switch. first working well, second no. If you interest i can send logs, or i can try fix it myself.

javicalle, 2024-02-11:
2 switches? Which names have they?
Are you sure that has been added to the `TuyaSingleSwitchDimmerGP` class?

closed 2024-08-17 by github-actions[bot]