zigpy/zha-device-handlers #2972: [Device Support Request] ZHA - Legrand / Bticino K4027C shutter doesn't support cover's position and status light

m4r1k, 2024-02-09:
### Problem description

### The problem

The `K4027C` BTicino switch in ZHA doesn't support three key features:
 - cover's `position`
 - `led_in_dark`
 - `led_if_on`
 - `identify`

Those are fully supported by zigbee2mqtt
https://www.zigbee2mqtt.io/devices/K4027C_L4027C_N4027C_NT4027C.html

Today I've confirmed that both cover's position and status light of the shutter cannot be configured with the ZHA integration.
While the status light is a nice to have, not supporting the position is a big missing feature, because the cover can only be either fully open or fully closed. In-between position have to be physically selected.

This issue was also reported on HA GitHub
https://github.com/home-assistant/core/issues/110008

### What version of Home Assistant Core has the issue?

core-2024.1.6

Zigbee Home Assistant (ZHA)

### Link to integration documentation on our website

https://www.home-assistant.io/integrations/zha

### Solution description

Support for the following 4 properties
 - cover's `position`
 - `led_in_dark`
 - `led_if_on`
 - `identify`

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

philippk75, 2024-02-12:
i would also need the support for bticino devices, at the moment i use it with zigbee2mqtt.

best regards philipp

shimon88, 2025-01-21:
FYI, I have the K4027C set using ZHA and after calibration the positioning is working ok.
what Is missing is the ability to set the cover type like we have in the Legrand app and the led control would be great.

shimon88, 2025-01-31:
@jafar-atili did you manage to calibrate this device while on Z3.0 network and not through Legrand gateway?

jafar-atili, 2025-01-31:
The K4027C can be calibrated by holding the up and down buttons for 5 seconds.

BTW, I don't have a legrand gateway.

I saw other calibration options on Z2M, and as mentioned before, due to the Zigbee 3.0 compliance on ZHA, the Bticino devices changes their behavior once they join a 3.0 gateway and some options will be limited.

I'd check the advanced calibration option offered in Z2M, you might find what you're looking for there.

shimon88, 2025-01-31:
I managed to calibrate using the up/down hold and just waiting for the calibration to take lace automatically but I could not perform the calibration as mentioned in Legrand forum. 
I quite like the ZHA integration over z2m so I think I will stick with it
thanks

jafar-atili, 2025-02-02:
> I managed to calibrate using the up/down hold and just waiting for the calibration to take lace automatically but I could not perform the calibration as mentioned in Legrand forum. 
> I quite like the ZHA integration over z2m so I think I will stick with it
> thanks

I agree. I run both ZHA and Z2M together on the same HA OS.

closed 2025-08-14 by github-actions[bot]