zigpy/zha-device-handlers #2973: [BUG] quirk (that removes level control) not loaded for all Tradfi plugs

erkr, 2024-02-10:
### Bug description

I have 3, IKEA tradfri plug outlets (relais on off). To my knowledge they are identical and have the same firmware.
I added them to ZHA long before there existed a quirk. Since the quirk was added to ZHA last year, only one of the plugs adopter the quirk (nicely not exposing level control anymore). But the other two plugs still run without the quirk.
Do I have to do something? 

### Steps to reproduce

N/A

### Expected behavior

I would expect the quirk loads for all, unless there is a device signature difference, that is not yet incorporated by the quirk

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x010a",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x1000",
        "0xfc7c"
      ],
      "output_clusters": [
        "0x0005",
        "0x0019",
        "0x0020",
        "0x1000"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [
        "0x0021"
      ],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "TRADFRI control outlet",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information
[... 648 lines cut ...]
        },
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

MattWestb, 2024-02-10:
With current production firmware and new added they is using version 2 of the quirk.
If being added with one older firmware is using version 1 or no quirk (i have one on current firmware and have being in the system for many years it was ZLL firmware and its not using any quirk).
If you like getting then quirked delete the device and wait one minute / restart HA and adding it new and it shall loading the V2 if being on current production firmware (endpoints is only read then adding devices new in the database and what the quirk is using for matching).

erkr, 2024-02-10:
@MattWestb thanks for you answer. I probably re-added the plug that is loading with the quirk for some reason. Can't recall that.
Really pity that re-adding is the only way. It could mess-up the mesh as they act as routers for more remote end points.

MattWestb, 2024-02-10:
I dont reparing then some end device (Aqara) is leaving the network if its parent is not online and its more important updating the firmware of the outlet then its some bad bugs in the old firmware that can killing the network.
Its shall being possible patching `zigbee.db` so its looks like the other outlets but its not recommended and not documented and the endpoints is only added then new paring so its little tricky to do it OK.
I recommending let tit if you can living with it and not getting other problems.

PS: I have IKEA Outlets as backbone in my Zigbee network and they is working good as long no device is rejoining the network to oft (parent accouterments bug in the Zigbee stack on old firmware) is also Aqara sensors is online for 3 years and only changing battery in them then needed.

MattWestb, 2024-02-10:
The endpoints and cluster for the with quirk and without and you see that some cluster is not the same but the "key" ones (On/Off) its the same so its working OK.
Both is on Firmware: 0x23089631 = current production version.
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x010a",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x1000"
      ],
      "output_clusters": [
        "0x0005",
        "0x0019",
        "0x0020",
        "0x1000"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [
        "0x0021"
      ],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "TRADFRI control outlet",
  "class": "plug.TradfriPlug2"
}

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x010a",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x1000",
        "0xfc7c"
      ],
      "output_clusters": [
        "0x0019",
        "0x0020",
        "0x1000"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "TRADFRI control outlet",
  "class": "zigpy.device.Device"
}
```
PS the quirk is changing the clusters for the report so must looking in the quirk for getting the real clusters.

erkr, 2024-02-11:
@MattWestb 
I will live with the level control (just hided them) to save me the trouble☺️
Mine are all version 0x20024623, according ZHA up te date 
Thanks for your help!

MattWestb, 2024-02-11:
Then you is having some strange settings in HA then its one old version.
Be sure not using the old very outdated test feed.
My config looks like this:
```
    ota:
      allow_file_providers: True
      otau_directory: /config/zigpy_ota
      ikea_provider: true
#      ikea_update_url: http://fw.test.ota.homesmart.ikea.net/feed/version_info.json
```
The last line is deactivated then its was the test feed and if you have it put one `#` or deleting the line and restart HA.

erkr, 2024-02-11:
I have `ikea_provider` false as I don't want to auto update my symfonisk devices.
Do you know if updates are manual now (as this setting enabled auto update for all ikea devices)

MattWestb, 2024-02-11:
You can downloading the OTA file for your device you like updating and having the IKEA OTA disables then it shall only looking for local files and updating. I dont knowing if the new GUI is disabling the auto update i have not testing it then all my devices is on latest version.
One PM: All IKEA controllers is blocked from firmware downgrade so id like going back is needing one SWD probe (J-Tag probe) and flashing one extracted OTA file on the chip.
All Routing device looks being not locked and can doing downgrading without problem only need parching ZHA for doing it (have doing some lights but i think the outlet is the same).

erkr, 2024-02-11:
Thanks for the advice. I only plan to update my IKEA plus, as I have my some Aqara sensors that are routed via them. These sensors are dropping off the network once in a while

erkr, 2024-02-11:
But to come back on to the original issue:
As a user I expect that when a quirk becomes available, it activates without the need to re-add the device.
I tried `reconfigure` in ZHA (and `misc_initialise` in the toolkit), without success.

erkr, 2024-02-11:
@MattWestb 
Sorry to bother again. I download the new firmware file for the IKEA Plug in the otau directory, and enabled the option in the configuration. It's not picked up by ZHA:
``` 

zha:
  zigpy_config:
    ota:
#      ikea_provider: true
#      ledvance_provider: true 
      otau_directory: /config/zigpy_ota/
  enable_quirks: true
  custom_quirks_path: /config/custom_quirks/
``` 

I wonder what the option `allow_file_providers:` does? It's not documented in ZHA. Same holds for `ikea_update_url:`.

MattWestb, 2024-02-11:
I was punting it in then the devs have blocking IKEA OTA but i dont knowing if its needed or not.
The PR is here for adding the option i have not seen any documentation https://github.com/zigpy/zigpy/pull/1337
Must restarting HA after doing the changes for it start being used.
The `ikea_update_url:` was made for IKEA test feed that was putting out beta firmware but some was bad and braking some controller and IKEA was putting in very old firmware in it and have not updating it over 3 years now.
ZHA is now using Dirigeras OTA feed that looks being the best then having more devices updated then the standard one for TF.
Some information if you like reading _little_ https://github.com/zigpy/zigpy/discussions/660

You is normally the loading in the (Z)HA log or put little debug loging and its writing all and look on the wiki for the options https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates

dmulcahey, 2024-02-11:
> I was punting it in then the devs have blocking IKEA OTA but i dont knowing if its needed or not. The PR is here for adding the option i have not seen any documentation [zigpy/zigpy#1337](https://github.com/zigpy/zigpy/pull/1337) Must restarting HA after doing the changes for it start being used. The `ikea_update_url:` was made for IKEA test feed that was putting out beta firmware but some was bad and braking some controller and IKEA was putting in very old firmware in it and have not updating it over 3 years now. ZHA is now using Dirigeras OTA feed that looks being the best then having more devices updated then the standard one for TF. Some information if you like reading _little_ [zigpy/zigpy#660](https://github.com/zigpy/zigpy/discussions/660)
> 
> You is normally the loading in the (Z)HA log or put little debug loging and its writing all and look on the wiki for the options https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates

We have it disabled by default because some devices can still be changed in ways users may not want. Like losing group binding. We had an issue w/ battery% too but that has since been fixed. We may enable this one by default in the next release. We’ll see…

MattWestb, 2024-02-11:
IKEA is not so bad now (the test feed was braking controllers but its long time ago) but the Z2M OTA files have some tuya and Aqara braking firmware (not implanted image number for the correct firmware) and some (TRVs) was very bad for the users.

erkr, 2024-02-12:
@MattWestb 
Thanks for the pointers, I'm not in a hurry to update the firmware, so I just wait until updating specific devices is settled.

@dmulcahey thanks for the great work. One question though;
What happens now when we enable the OTA providers? Will it still enable auto update (as documented), OR provide manual control to either update a device or skip the updates?

TheJulianJES, 2024-02-21:
Just as a note, this original issue is still present.
We just need to also add a quirk for that signature (which can be generated using [quirk-generator](https://github.com/zigpy/quirk-generator) btw) or wait for "quirks v2" that can replace multiple quirks for slightly different signatures.

MattWestb, 2024-02-21:
If adding one outlet new on current production `Firmware: 0x23089631` its using `Quirk: plug.TradfriPlug2` and shall being OK.
If the device was added with one older firmware and being update you need deleting  it and wait one minute / restart HA and adding it new and it shall working OK.

erkr, 2024-02-21:
@TheJulianJES 
Any indication when quirk v2 is in the release? I don't have a burning issue with these plugs, so  I prefer to wait.
Best Eric

erkr, 2024-06-23:
Today I noticed the 3 ikea plugs where running on a new v2 quirk ( zigpy.quirks.v2.CustomDeviceV2 ) that first expose dimming entities for these on/off plugs 🥳

closed 2024-06-23 by erkr