zigpy/zha-device-handlers #2974: [Device Support Request]  TZE204_ztqnh5cg

PieterHattinghZA, 2024-02-10:
### Problem description

ZHA picks up the device but as a Smart Plug and only shows the Firmware status. I tried to use a quirk as exampled in various posts.

I added the TZE204_ztqnh5cg entry under the models sections, configured my quirks path and file and zero luck.

Could someone please assist me. I am not a coder by any stretch of the imagination. 

[Diagnostics.json](https://github.com/zigpy/zha-device-handlers/files/14230604/zha-4008cfd83e4f39027d6b9c5120460dff-_TZE204_ztqnh5cg.TS0601-19f0463e15f2075c2f1a29e820082c90.json)

Configuration.yaml entry

zha:
  enable_quirks: true
  custom_quirks_path: /config/custom_zha_quirks/
  database_path:  /config/zigbee.db
  

### Solution description

I would appreciate a custom quirk and some guidance in using/applying it

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

As per the screenshot, the custom quirk is not used to id the device. HELP

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_ztqnh5cg",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[custom_zha_quirks.zip](https://github.com/zigpy/zha-device-handlers/files/14230615/custom_zha_quirks.zip)
```

</details>

vshijiav, 2024-02-17:
"""ZY-M100 Human Presence Sensor"""
import math
from typing import Dict, Optional, Tuple, Union

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    AnalogInput,
    AnalogOutput,
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.measurement import (
    IlluminanceMeasurement,
    OccupancySensing,
    RelativeHumidity,
    TemperatureMeasurement,
)
from zigpy.zcl.clusters.security import IasZone

from zhaquirks import Bus, LocalDataCluster, MotionOnEvent
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    MOTION_EVENT,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    #DPToAttributeMapping,
    TuyaLocalCluster,
    TuyaManufCluster,
    TuyaNewManufCluster,
    TuyaDPType,

)

from zhaquirks.tuya.mcu import (
    TuyaAttributesCluster,
    DPToAttributeMapping,
    TuyaMCUCluster,
)

ZONE_TYPE = 0x0001

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaAnalogInput(AnalogInput, TuyaLocalCluster):
    """Tuya local AnalogInput cluster."""

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

class TuyaTemperatureMeasurement(TemperatureMeasurement, TuyaLocalCluster):
    """Tuya local TemperatureMeasurement cluster."""

class TuyaRelativeHumidity(RelativeHumidity, TuyaLocalCluster):
    """Tuya local RelativeHumidity cluster."""

class MmwRadarManufCluster(TuyaMCUCluster):
[... 257 lines cut ...]
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarFadingTime,
                ],
                OUTPUT_CLUSTERS: [],
            },
            6: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarSensitivity,
                ],
                OUTPUT_CLUSTERS: [],
            },
			#### end of new replacements.
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
	
You can use this for now, at least it shows occupancy and some attributes, but clearly some of the stuff is not right.... Tried modify according to zigbee2mqtt file it still no luck for the settings and lux attribute

PieterHattinghZA, 2024-02-19:
Thanks.

I tried the quirk, but still not working

Regards,

[signatureImage]

This e-mail and any attachments thereto are strictly confidential and are intended solely for the use of the addressee. Should you not be the intended addressee, please delete this e-mail message immediately. While care is taken in preparing this document, no representation, warranty or undertaking (expressly or implied) is given and no responsibility nor liability is accepted by the company as to the accuracy of the information contained herein, that the email is free of viruses, or for any damages that may occur from receiving or opening this email.

vshijiav, 2024-02-19:
umm---pretty strange, I actually modified another version of the quirk according to zigbee2mqtt. They have this TZE204_ztqnh5cg supported and their DP point is            
            [1, 'presence', tuya.valueConverter.trueFalse1],
            [9, 'target_distance', tuya.valueConverter.divideBy100],
            [104, 'illuminance_lux', tuya.valueConverter.raw],
            [2, 'radar_sensitivity', tuya.valueConverter.raw],
            [4, 'maximum_range', tuya.valueConverter.divideBy100],
            [3, 'minimum_range', tuya.valueConverter.divideBy100],
            [102, 'fading_time', tuya.valueConverter.divideBy10],
            [101, 'detection_delay', tuya.valueConverter.divideBy10],
But I tried and still no lux reported. very strange, Im no quirk expert and hope someone could figure this out soon...

PieterHattinghZA, 2024-02-19:
I think my problem is that the quirk is not used.

Really a noob at ZHA, and quirks are foreign to me.

Regards,

[signatureImage]

This e-mail and any attachments thereto are strictly confidential and are intended solely for the use of the addressee. Should you not be the intended addressee, please delete this e-mail message immediately. While care is taken in preparing this document, no representation, warranty or undertaking (expressly or implied) is given and no responsibility nor liability is accepted by the company as to the accuracy of the information contained herein, that the email is free of viruses, or for any damages that may occur from receiving or opening this email.

vshijiav, 2024-02-19:
you will have to add something like this in configuration.yaml first
zha:
  custom_quirks_path: /config/zha_quirks/

and then upload quirks into zha_quirks folder, reload, done.

PieterHattinghZA, 2024-02-19:
I've done all that.

Also cleared the cache etc

Regards,

[signatureImage]

This e-mail and any attachments thereto are strictly confidential and are intended solely for the use of the addressee. Should you not be the intended addressee, please delete this e-mail message immediately. While care is taken in preparing this document, no representation, warranty or undertaking (expressly or implied) is given and no responsibility nor liability is accepted by the company as to the accuracy of the information contained herein, that the email is free of viruses, or for any damages that may occur from receiving or opening this email.

vshijiav, 2024-02-23:
> I've done all that. Also cleared the cache etc Regards, [signatureImage] This e-mail and any attachments thereto are strictly confidential and are intended solely for the use of the addressee. [...]

[image]

I just found that my newly edited quirk seems working with all options showed up... so that DP point from zigbee2mqtt is actually good.

PieterHattinghZA, 2024-02-24:
That is awesome

Please share with me

A64F57C3-565A-4955-AC58-BB74B327503F.png (view on web)<https://github.com/zigpy/zha-device-handlers/assets/27871589/2abe7177-1233-433d-8dbe-60a1d52fbe45>

vshijiav, 2024-02-27:
[test.zip](https://github.com/zigpy/zha-device-handlers/files/14414650/test.zip)

PieterHattinghZA, 2024-02-29:
I tested your file on my system and now it pics up more entities.

I will send you the one that provided me with Illumanance and motion detection.

Maybe with the two files combined, we can get further

vshijiav, 2024-02-29:
> I tested your file on my system and now it pics up more entities. I will send you the one that provided me with Illumanance and motion detection. Maybe with the two files combined, we can get further
> […](#)

My newly uploaded file is actually a newly modded version and it reports lux, but it just takes some time to start reporting.
[image]

illigtr, 2024-03-14:
Hi, I have this sensor as well. Your last post suggested a merged file with controls and sensors that is working. Can you share?

vshijiav, 2024-03-14:
> Hi, I have this sensor as well. Your last post suggested a merged file with controls and sensors that is working. Can you share?

> [test.zip](https://github.com/zigpy/zha-device-handlers/files/14414650/test.zip)

This file is the one you need

illigtr, 2024-03-14:
Thanks, but this file does not expose the lux sensor… it is the same file
as mentioned in the thread…

*From:* vshijiav ***@***.***
*Sent:* March 14, 2024 15:35
*To:* zigpy/zha-device-handlers ***@***.***>
*Cc:* illigtr ***@***.***>; Comment ***@***.***>
*Subject:* Re: [zigpy/zha-device-handlers] [Device Support Request]
TZE204_ztqnh5cg (Issue #2974)

Hi, I have this sensor as well. Your last post suggested a merged file with
controls and sensors that is working. Can you share?

test.zip
<https://github.com/zigpy/zha-device-handlers/files/14414650/test.zip>

This file is the one you need

vshijiav, 2024-03-14:
you will have to wait for 1 day or so. The lux will come out

vshijiav, 2024-03-15:
> No change, the lux entity simply doesn't even exist in the GUI.
> 
> However, I was able to use this quirk `zhaquirks/tuya/ts0601_motion.py` successfully by simply adding the correct device string to the list of supported devices. Now the lux entity appears immediately and works.
> 
> Found the answer here: [https://github.com/zigpy/zha-device-handlers/pull/1928](url)

Pretty Strange I checked the file I provided it is the same as the one Im using, and it shows lux in my system. Not sure what is happening. And I take a look at the file you provided it is the first quirk I tried (I added model number as well), and it does not work on my end. So I went to zigbee2mqtt side and modified my own quirk and it is now working. Can I have a look at your product picture? I suspect it is not the same product but share a same SKU.

illigtr, 2024-03-15:
[https://www.aliexpress.com/item/1005006374353536.html](url)
[image]

vshijiav, 2024-03-15:
> [https://www.aliexpress.com/item/1005006374353536.html](url) [...]

I can confirm this is a completely different device from the one in my hand. So it is same model number but different device.

closed 2024-09-19 by github-actions[bot]