zigpy/zha-device-handlers #2977: ZHA battery doubling on some IKEA devices still present

ProfDrYoMan, 2024-02-12:
And which device is affected?

_Originally posted by @TheJulianJES in https://github.com/zigpy/zha-device-handlers/issues/2948#issuecomment-1938164896_

[image]

I have multiple of them (old and recently bought) . For some the battery seems correct (but not sure, because the battery is old in there).

No idea what `sw_build_id` is for these devices. Diagnosis data will follow this eventing.

TheJulianJES, 2024-02-12:
Thanks! Yeah, the diagnostic information should contain the `sw_build_id` (if it was read once).
(It can be downloaded on that page, three dots on the left, download device diagnostics.)

ProfDrYoMan, 2024-02-12:
`"0x4000": {
                "attribute_name": "sw_build_id",
                "value": "2.4.5"
              }`

[zha-6a619c3d26dd0055dbea11733a550770-IKEA of Sweden Remote Control N2-c4e1da74fe0a76be750e35dda7bc907c.zip](https://github.com/zigpy/zha-device-handlers/files/14235966/zha-6a619c3d26dd0055dbea11733a550770-IKEA.of.Sweden.Remote.Control.N2-c4e1da74fe0a76be750e35dda7bc907c.zip)

Neither of the two which you asked on top. :)

ProfDrYoMan, 2024-02-12:
[zha-6a619c3d26dd0055dbea11733a550770-IKEA of Sweden Remote Control N2-fa2643eb5d69a235d7e86b707f5c2d77.zip](https://github.com/zigpy/zha-device-handlers/files/14235982/zha-6a619c3d26dd0055dbea11733a550770-IKEA.of.Sweden.Remote.Control.N2-fa2643eb5d69a235d7e86b707f5c2d77.zip)

And all other same type buttons. Some of them showing < 100%, but not sure if battery is half empty.

ProfDrYoMan, 2024-02-12:
Ok, is this FW newer or older? I have updates enabled in ZHA in HA via config.

```zha:
  zigpy_config:
    ota:
      ikea_provider: true
      ledvance_provider: true
      salus_provider: true
      inovelli_provider: true
      thirdreality_provider: true```

PS: where are the OTA files stored, if there is no config path set?

MattWestb, 2024-02-12:
I have one that is in RCP test system but is sending 255 as battery remainig and also voltage = not valid so its device problem / battery and using quirk V2 with alkaline batteries and one with Ni-MH one and is working OK. Putting in new Mi-MH and its working OK.
In production i have 3 and all is showing 200% and is using quirk V1 and is not using the quirk fix of some reason.
```

  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0820",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x1000",
        "0xfc57"
      ],
      "output_clusters": [
        "0x0003",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "Remote Control N2",
  "class": "zhaquirks.ikea.fourbtnremote.IkeaTradfriRemoteV1"
}
```
Very likely all production controller was installed with V1 firmware and have being updated but the quirk is not being updated then the cluster settings is not updated in ZHA.
Is it one easy way getting the endpoints reading the cluster settings so its loading the  right quirk without adding in new ?
The missing is cluster "0xfc7c" for V2 to being loaded and fixing the battery readings.

MattWestb, 2024-02-12:
@ProfDrYoMan  In one of the diagnostic its running 2.4.5 = the latest production version but is still using quirk V1 as my production is doing

ProfDrYoMan, 2024-02-12:
Matt, I don't get if this is an ask or just an information into my direction. I am not into any details, I just provided my feedback. 

If I get clear asks on what to further provide, I will try to do so.

Info: The different devices are ~1 year different in age. That means time when I bought them.

MattWestb, 2024-02-12:
Your devices is running the latest production firmware = OK
ZHA is still using the quirk for the fabric firmware (V1) like all my production is using.
If you like you can deleting one controller wait one minute (or restart HA) and adding it new then it shall using the quirk V2 and doing the battery OK but you is loosing the binding if you have doing it.

ProfDrYoMan, 2024-02-12:
Now I understand. Let's see, can do that for the first "offending" device in the evening.

Is there a way or plan to fix that the v2 version of the quirk is used without going thru a new binding process?

MattWestb, 2024-02-12:
The V2 is having one extra cluster that can being patched in Zigbee.db but i not dont  doing it on my production system but its possible adding it for all devices and then ZHA shall loading the V2 quirk.
Alternative doing one local quirk with V2 class with V1 signature shall also working but its little wrong doing it.

MattWestb, 2024-02-12:
@TheJulianJES What about doing one new device class V1.5 for updated device using old cluster settings and using your doubling logic if SW >2.4 not doubling ?

ProfDrYoMan, 2024-02-12:
Remove, Wait 5 min+, Relearn. Did have battery removed.

[image]

Clearly v2 quirk now.

But: ~5 min and no updates on battery status yet.

[zha-6a619c3d26dd0055dbea11733a550770-IKEA of Sweden Remote Control N2-c4e1da74fe0a76be750e35dda7bc907c.zip](https://github.com/zigpy/zha-device-handlers/files/14241429/zha-6a619c3d26dd0055dbea11733a550770-IKEA.of.Sweden.Remote.Control.N2-c4e1da74fe0a76be750e35dda7bc907c.zip)

MattWestb, 2024-02-12:
Its expected IKEA controllers is reporting battery long between.
For getting somthing read the Attributes of the cluster
`battery_percentage_remaining (id: 0x0021)` on 
Clusters` PowerConfiguration` (Endpoint id: 1, Id: 0x0001, Type: in) and waking the device up then sending to it.

Also try do one reconfigure from the device card ao the reporting is being OK.

ProfDrYoMan, 2024-02-12:
Reading `battery_percentage_remaining (id: 0x0021)` with wakeup button press did work. Gives 255.

Reconfigure (with wakeup):
[image]

All 4 Buttons work, but no battery status.

MattWestb, 2024-02-12:
You is getting 255 = not valid then using good alkaline batteries but its going away then they is going down. Try putting in rechargeable one and see if its being OK (i was doing it yesterday on one and it was working OK).

MattWestb, 2024-02-12:
then all is looking good but IKEA is having one bug i the battery voltage then i think you have charging them so hard so thy is delivering 25.5 V :-)
The type and quantity i have making looking nice in the quirk then they is not reporting it.

ProfDrYoMan, 2024-02-12:
> The type and quantity i have making looking nice in the quirk then they is not reporting it.

The same type device shows it in quirkv1 but not in quirkv2?

RSZEE, 2024-02-18:
## FIX FOUND
### (May Work For You, Worked For Me)

1. Press and Hold the device pairing button for 10s
2. Remove 1 battery and do step 1
3. Remove the second battery and do step 1
4. Restart Home Assistant
5. Temporarily Disable Debug Logging (If you have it on)
6. Re Add The Device (4 Presses On The Pairing Button)
7. Wait 5 minutes
8. Go To Manage Device -> Attributes -> PowerConfiguration -> battery_remaining_percentage
9. Wake the device via spam pressing one of the buttons
10. Read the attribute
11. Make sure the attribute is 200 or under
12. Fixed

RSZEE, 2024-02-18:
Remove From Home Assistant In Step 2 aswell

RSZEE, 2024-02-18:
Worked With IKEA LADDA Rechargable Batteries

MattWestb, 2024-02-18:
One warming for point 1 its very likely kidnapping one light then its doing Light Link Paring. 
But the good is the device is doing one reset and losing the network (was having problem doing one direct pairing with one OnOff button after upgrade nut it was working with normal EZ pairing).

The important is ping 2 and 3 and then ZHA is pairing it new and using the new setting without double battery renaming.

PS: You can still getting 255 in the reading = not valid, also with rechargeable batteries but is normally being to high voltages and its being fixed then they is going down.

TheJulianJES, 2024-02-21:
Another note (for me). The `replacement` for the `IkeaTradfriRemoteV2` remote should have the `PowerConfig2AAACluster` cluster to show more information for the battery entity extra attributes.

MattWestb, 2024-02-21:
Styrbad was having wrong metadata in the TF feed but is OK in Dirigera but i was updating it with override / patched metadata but 95% of the users was not getting there remotes updated.

I think the extended battery information was missing then the V2 was being added and i was not double cheeking it but it was added for V1 and also for the new 1xAAA devices at the same time.

closed 2024-02-23 by TheJulianJES