zigpy/zha-device-handlers #2980: [Device Support Request] TS0601_TZE200_r32ctezx Lerlink (Tuya) Fan Switch (5 speed)

demushkin, 2024-02-13:
### Problem description

Device connected without items.
But with ts0601_dimmer.TuyaSingleSwitchDimmerGP it works only as switch (on/off) and not support levels (for fan control)
Device is already supported with z2m https://github.com/Koenkk/zigbee2mqtt/issues/16748

### Solution description

support fan control (5 levels)

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_r32ctezx",
  "model": "TS0601",
  "class": "ts0601_dimmer.TuyaSingleSwitchDimmerGP"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2024.2.1",
    "dev": false,
    "hassio": true,
[... 377 lines cut ...]
Last logged: 18:14:55

Loaded custom quirks. Please contribute them to https://github.com/zigpy/zha-device-handlers
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
class TuyaSingleSwitchDimmerGP(TuyaDimmerSwitch):
    """Tuya touch switch device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_3p5ydos3", "TS0601"),
            ("_TZE200_ip2akl4w", "TS0601"),
            ("_TZE200_vucankjx", "TS0601"),  # Loratap
            ("_TZE200_y8yjulon", "TS0601"),
            ("_TZE200_r32ctezx", "TS0601"),
```

</details>

demushkin, 2024-08-17:
> There hasn't been any activity on this issue recently. Due to the high number of incoming GitHub notifications, we have to clean some of the old issues, as many of them have already been resolved [...]

still not supported as fan speed control

Karakandass, 2025-01-09:
I bought same FAN controller and it would be nice if the device will have support in ZHA. Is it possible?

demushkin, 2025-01-10:
You can use this custom quirk
https://github.com/demushkin/ha/blob/main/custom_zha_quirks/TUYA_LERLINK_FAN_TS0601_TZE200_r32ctezx.py
it support on/off only, not fan speed in zha now ((

closed 2025-08-14 by github-actions[bot]