zigpy/zha-device-handlers #2987: [Device Support Request] HZC Electronics D688-ZG

EmberLightVFX, 2024-02-17:
### Problem description

I’m having a problem when I try to toggle my light dimmer in the UI.
The dimmer is a HZC Electronics D688-ZG dimmer.
When I press to turn the light off, the light turns off but it doen’t seem to register in the UI and the switch goes back to its on-state in the UI while the real light is still off.

If I turn on/off the light with the light-switch, HA will register this and print it in the log book

If I go to the switch’s Manage Zigbee Device I can send the light-toggle from there and it works as expected. I do get an error: Failed to call service zha/issue_zigbee_cluster_command. Failed to issue cluster command with status: <Status.UNSUP_CLUSTER_COMMAND: 129> but the light works as expected and it’s being logged in the logbook.

It might be connected to this support request also as they are both made by HZC: https://github.com/zigpy/zha-device-handlers/issues/2865

### Solution description

The light should turn on/off and stay that way after changing it in the UI.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4714, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0101",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0b05",
        "0x1000"
      ],
      "output_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "Shyugj",
  "model": "Dimmer-Switch-ZB3.0",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

[... 591 lines cut ...]
2024-02-17 14:29:34.440 DEBUG (Thread-3) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f50d6f197b0>, '\n            INSERT INTO attributes_cache_v12\n            VALUES (:ieee, :endpoint_id, :cluster_id, :attrid, :value, :timestamp)\n                ON CONFLICT (ieee, endpoint_id, cluster, attrid) DO UPDATE\n                SET value=excluded.value, last_updated=excluded.last_updated\n                WHERE\n                    value != excluded.value\n                    OR :timestamp - last_updated > :min_update_delta\n            ', {'ieee': a4:c1:38:1c:52:fe:e4:1c, 'endpoint_id': 1, 'cluster_id': 2820, 'attrid': 1291, 'value': 205, 'timestamp': 1708176574.435183, 'min_update_delta': 30.0}) completed
2024-02-17 14:29:34.440 DEBUG (Thread-3) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f50d6f197b0>)
2024-02-17 14:29:34.441 DEBUG (Thread-3) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f50d6f197b0>) completed
2024-02-17 14:29:34.442 DEBUG (Thread-3) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f50d6f197b0>, '\n            INSERT INTO attributes_cache_v12\n            VALUES (:ieee, :endpoint_id, :cluster_id, :attrid, :value, :timestamp)\n                ON CONFLICT (ieee, endpoint_id, cluster, attrid) DO UPDATE\n                SET value=excluded.value, last_updated=excluded.last_updated\n                WHERE\n                    value != excluded.value\n                    OR :timestamp - last_updated > :min_update_delta\n            ', {'ieee': a4:c1:38:1c:52:fe:e4:1c, 'endpoint_id': 1, 'cluster_id': 2820, 'attrid': 1288, 'value': 941, 'timestamp': 1708176574.435911, 'min_update_delta': 30.0})
2024-02-17 14:29:34.443 DEBUG (Thread-3) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f50d6f197b0>, '\n            INSERT INTO attributes_cache_v12\n            VALUES (:ieee, :endpoint_id, :cluster_id, :attrid, :value, :timestamp)\n                ON CONFLICT (ieee, endpoint_id, cluster, attrid) DO UPDATE\n                SET value=excluded.value, last_updated=excluded.last_updated\n                WHERE\n                    value != excluded.value\n                    OR :timestamp - last_updated > :min_update_delta\n            ', {'ieee': a4:c1:38:1c:52:fe:e4:1c, 'endpoint_id': 1, 'cluster_id': 2820, 'attrid': 1288, 'value': 941, 'timestamp': 1708176574.435911, 'min_update_delta': 30.0}) completed
2024-02-17 14:29:34.443 DEBUG (Thread-3) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f50d6f197b0>)
2024-02-17 14:29:34.443 DEBUG (Thread-3) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f50d6f197b0>) completed
2024-02-17 14:29:34.444 DEBUG (Thread-3) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f50d6f197b0>, '\n            INSERT INTO attributes_cache_v12\n            VALUES (:ieee, :endpoint_id, :cluster_id, :attrid, :value, :timestamp)\n                ON CONFLICT (ieee, endpoint_id, cluster, attrid) DO UPDATE\n                SET value=excluded.value, last_updated=excluded.last_updated\n                WHERE\n                    value != excluded.value\n                    OR :timestamp - last_updated > :min_update_delta\n            ', {'ieee': a4:c1:38:1c:52:fe:e4:1c, 'endpoint_id': 1, 'cluster_id': 2820, 'attrid': 1285, 'value': 229, 'timestamp': 1708176574.436346, 'min_update_delta': 30.0})
2024-02-17 14:29:34.444 DEBUG (Thread-3) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f50d6f197b0>, '\n            INSERT INTO attributes_cache_v12\n            VALUES (:ieee, :endpoint_id, :cluster_id, :attrid, :value, :timestamp)\n                ON CONFLICT (ieee, endpoint_id, cluster, attrid) DO UPDATE\n                SET value=excluded.value, last_updated=excluded.last_updated\n                WHERE\n                    value != excluded.value\n                    OR :timestamp - last_updated > :min_update_delta\n            ', {'ieee': a4:c1:38:1c:52:fe:e4:1c, 'endpoint_id': 1, 'cluster_id': 2820, 'attrid': 1285, 'value': 229, 'timestamp': 1708176574.436346, 'min_update_delta': 30.0}) completed
2024-02-17 14:29:34.446 DEBUG (Thread-3) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f50d6f197b0>)
2024-02-17 14:29:34.447 DEBUG (Thread-3) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f50d6f197b0>) completed

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

EmberLightVFX, 2024-03-03:
Made a custom quirk that seems to fix my problem. I simply ignore the response commands from the OnOff cluster (I think).
The identification still doesn't work and I couldn't get that working.
I also don't know how to correctly do a PR out of this but it's at least something.

jagheterfredrik, 2024-09-14:
@EmberLightVFX would you mind sharing?

EmberLightVFX, 2024-09-14:
@jagheterfredrik You'll find the custom quirk under the `Custom quirk` dropdown in the first post

jagheterfredrik, 2024-09-14:
Ah cheers @EmberLightVFX, I'll make a PR

closed 2024-09-24 by TheJulianJES

Pieter8601, 2025-04-24:
Hi all, i have been looking through these discussions because i have the same problem with a eco dim 05, i understand there is a solution by changing the quirk but could someone please guide me to how i should change this step by step. i'm pretty new to HA and zigbee.

PR #3362 (2024-09-24): Add HZC D688-ZG dimmer switch quirk
## Proposed change

Add quirk for HZC Dimmer-Switch-ZB3.0 Rotary Dimmer. Shout out to @EmberLightVFX for making the quirk.

## Additional information

Fixes #2987

## Checklist

- [X] The changes are tested and work correctly
- [X] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works