zigpy/zha-device-handlers #2988: [BUG] STYRBAR REMOTE not showing battery percentage

RSZEE, 2024-02-17:
### Bug description

My STYRBAR remote has had issues recently with showing its battery percentage. When I first paired it it was showing, then it disconnected. When I repaired it it showed the battery percentage as 'Unknown'. It has been a few weeks and it still hasen't showed up.
I have:
- Unpaired and repaired
- Removing and adding the batteries back in
- Reconfiguring

### Steps to reproduce

Go to Devices -> ZHA -> STYRBAR REMOTE -> Battery

### Expected behavior

To show the battery percentage.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]
[image]
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0820",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x1000",
        "0xfc57",
        "0xfc7c"
      ],
      "output_clusters": [
        "0x0003",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "Remote Control N2",
  "class": "zhaquirks.ikea.fourbtnremote.IkeaTradfriRemoteV2"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Additional information

Started after repairing.

cjcfreitas, 2024-02-21:
I have two of these devices. None shows the battery level (and, I think, they never did). Reports "unknown". However, the Rodret device (simpler two button Ikea remote) displays the battery level.

RSZEE, 2024-02-21:
## FIX FOUND
### (May Work For You, Worked For Me)

1. Press and Hold the device pairing button for 10s
2. Remove 1 battery and do step 1
3. Remove the second battery and do step 1
4. Restart Home Assistant
5. Temporarily Disable Debug Logging (If you have it on)
6. Re Add The Device (4 Presses On The Pairing Button)
7. Wait 5 minutes
8. Go To Manage Device -> Attributes -> PowerConfiguration -> battery_remaining_percentage
9. Wake the device via spam pressing one of the buttons
10. Read the attribute
11. Make sure the attribute is 200 or under
12. Fixed

RSZEE, 2024-02-21:
Remove From ZHA In Step **2**
Used **IKEA LADDA Rechargeable Batteries**

RSZEE, 2024-02-21:
Make sure you are **Far** away from a IKEA Light Bulb as it might connect to it instead.

RSZEE, 2024-02-21:
[image]

RSZEE, 2024-02-21:
On a different bug report the creator of the ZHA Quirks has said that there is an issue with the quirk.

closed 2024-08-26 by github-actions[bot]

miquelblanch, 2024-09-06:
Hello,

I have the same bug on showing the battery percentage on the Styrbar Remote Control. I've tried the solution below:

https://github.com/zigpy/zha-device-handlers/issues/2988#issuecomment-1957458093

but I couldn't resolve it.

What do you mean by "Go To Manage Device -> Attributes -> PowerConfiguration -> battery_remaining_percentage"? am I able to modify the Attributes of the device? Are LADDA rechargeable batteries necessary? May you please explain the whole process?

Thanks in advance!!

MattWestb, 2024-09-06:
If using alkaline battier they is 1.5 V and rechargeable is 1.2 V the remote is sending 255 = not valid  / cutoff range reading and ZHA is showing not avable reading unit the  battery is going down after some mounts.

miquelblanch, 2024-09-06:
ok thanks for the response,... my problem is that doesn't work neither with normal batteries nor with LADDA rechargeables...

KipiKland, 2024-09-29:
> If using alkaline battier they is 1.5 V and rechargeable is 1.2 V the remote is sending 255 = not valid / cutoff range reading and ZHA is showing not avable reading unit the battery is going down after some mounts.

Correct, a full charged LADA battery will provide 1.35 volt max, that is around 100% charge, evrything over will trigger a NULL value on percentage on "battery" value

miquelblanch, 2024-09-29:
Ok perfecte, thanks for the answer,... I'll use LADA battery.