zigpy/zha-device-handlers #2994: [BUG] Aqara H1 EU - Incorrect temperature value

Mookunicorn, 2024-02-20:
### Bug description

Hello everyone !

On the device "lumi.switch.l2aeu1" ("Aqara H1 EU") the temperature indicated is 0.2 which is not the right value at all. On the cluster the temperature displayed is indeed 17.

Any idea how to fix this?

### Steps to reproduce

Settings > Device and Services > Zigbee Home Automation > Select "lumi.switch.l2aeu1" called "Interrupteur chambre" > Show Show the temperature sensor & show the temperature in cluster

### Expected behavior

The temperature must be the same as that indicated by the cluster

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0xfcc0"
      ],
      "output_clusters": []
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
[... 544 lines cut ...]
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

[home-assistant_zha_2024-02-20T17-01-58.450Z.log](https://github.com/zigpy/zha-device-handlers/files/14348943/home-assistant_zha_2024-02-20T17-01-58.450Z.log)

</details>

TheJulianJES, 2024-02-20:
"DeviceTemperature" probably isn't supported on that device then. Some Aqara devices still expose it for some reason.
We could just remove it.

Mookunicorn, 2024-02-20:
It's a bit surprising since the correct temperature was displayed before I restarted Home Assistant :/

[image]

I guess I'll have to resolve to remove this sensor.

Mookunicorn, 2024-02-20:
> "DeviceTemperature" probably isn't supported on that device then. Some Aqara devices still expose it for some reason. We could just remove it.

It seems that after yet another reset of the device, well it works again, I even have new sensors which are "unknown" (I would delete those). I noticed that a Quirks is added to the device, perhaps this is the solution to the problem.

In any case, I thank you for your strength which allowed everything to visibly work.

[image]

TheJulianJES, 2024-02-20:
Oh hmm, it might be an issue that the quirk didn't apply before for some reason.
I'll leave this open for now, so I don't forget about it.

closed 2024-08-25 by github-actions[bot]