zigpy/zha-device-handlers #2999: [Device Support Request] for TS0601 TZE204_5cuocqty (avatto 1 ch zigbee dimmer)

vrtsman, 2024-02-24:
### Problem description

device entities not shown / discoverd, just the firmware version

### Solution description

please add this device (also the 2 way dimmer device( to the quirks or full ZHA supported devices

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

Suxsem, 2024-04-13:
datapoints found until now:

            [1, 'state', 
            [2, 'brightness', 
            [???, 'min_brightness', 
            [4, 'switch_type',
            [???, 'max_brightness',
            [6, 'countdown',
            [14, 'power_on_behavior',
            [???, 'light_type',

cannot find min_brightness, max_brightness and light_type

simoesp, 2024-04-24:
+1 Diagnostic Upload
[zha-1b1c0fd13410c1aebdc6ad142bff5c39-_TZE204_5cuocqty TS0601-22d02299e374c3016a255dd60f89b131.json](https://github.com/zigpy/zha-device-handlers/files/15096369/zha-1b1c0fd13410c1aebdc6ad142bff5c39-_TZE204_5cuocqty.TS0601-22d02299e374c3016a255dd60f89b131.json)

stavragi, 2024-06-10:
I was watching this thread for updates since i have the same device so i thought i should inform everyone here.
It seems someone managed to get it working:
https://github.com/zigpy/zha-device-handlers/issues/3141

It doesn't seem to pulled to the main branch yet, but I tested the suggested changes and it worked for me.

closed 2024-12-14 by github-actions[bot]