zigpy/zha-device-handlers #3001: [Device Support Request] TS011F / _TZ3000_okaz9tjs recognized as light?

Commifreak, 2024-02-25:
### Problem description

After binding the plug, HA shows its switch as light.

### Solution description

Detection as switch?

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4660, maximum_buffer_size=108, maximum_incoming_transfer_size=0, server_mask=11264, maximum_outgoing_transfer_size=0, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000a",
        "0x0702",
        "0x0b04",
        "0x1000",
        "0xe000",
        "0xe001"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "_TZ3000_okaz9tjs",
  "model": "TS011F",
  "class": "zigpy.device.Device"
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

HA 2024.2.3

GuidoC1, 2024-03-08:
@Commifreak , I have the same device and applied the quirk provided in [this post](https://github.com/zigpy/zha-device-handlers/issues/1764#issuecomment-1946842958). It works fine for me and multiple other users. Maybe you can give the custom quirk provided there a try

Commifreak, 2024-04-21:
How can this then be included in ZHA quirks collection?