zigpy/zha-device-handlers #3015: [Device Support Request]

gijsh070, 2024-03-02:
### Problem description

Hello,

I bought a CO + CH4 detector. I am trying to get this configured in ZHA with a custom quirk but it seems to mismatch or give error ImportError: attempted relative import with no known parent package.

I am kind of lost in this and hope anyone can suggest some troubleshooting i could do. I see that this device can be converted in zigbee2mqtt but i do not want to move from ZHA to zigbee2mqtt. I have tried different kind of quirks, with the latest one the Moes TS0601.

TS0601
TZE200_iuk8kupi

### Solution description

Troubleshooting tips or custom quirk for this device. Any help would be highly appreciated.

### Screenshots/Video

[image]

### Device signature

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_iuk8kupi",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

"""Gas Sensor."""
import logging

import zigpy.profiles.zha
from zigpy.quirks import CustomCluster, CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.security import IasZone

from zhaquirks import Bus
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    ZONE_STATUS,
    ZONE_TYPE,
)

from . import TuyaManufCluster, TuyaManufClusterAttributes

_LOGGER = logging.getLogger(__name__)

TUYA_GAS_DETECTED_ATTR = 0x0401  # [0]/[1] [Detected]/[Clear]!

class TuyaGasDetectorCluster(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of the TS0601 gas detector."""

    attributes = TuyaManufClusterAttributes.attributes.copy()
    attributes.update(
        {
            TUYA_GAS_DETECTED_ATTR: ("gas_detected", t.uint8_t, True),
        }
    )

    def _update_attribute(self, attrid, value):
        super()._update_attribute(attrid, value)
        if attrid == TUYA_GAS_DETECTED_ATTR:
            if value == 0:
                self.endpoint.device.ias_bus.listener_event(
                    "update_zone_status", IasZone.ZoneStatus.Alarm_1
                )
            else:
                self.endpoint.device.ias_bus.listener_event("update_zone_status", 0)
        else:
            _LOGGER.warning(
                "[0x%04x:%s:0x%04x] unhandled attribute: 0x%04x",
                self.endpoint.device.nwk,
                self.endpoint.endpoint_id,
                self.cluster_id,
                attrid,
            )

class TuyaGasDetectorZone(CustomCluster, IasZone):
    """IAS Zone."""

    _CONSTANT_ATTRIBUTES = {ZONE_TYPE: IasZone.ZoneType.Carbon_Monoxide_Sensor}

    def __init__(self, *args, **kwargs):
        """Init."""
        super().__init__(*args, **kwargs)
        self.endpoint.device.ias_bus.add_listener(self)

    def update_zone_status(self, value):
        """Update IAS status."""
        super()._update_attribute(ZONE_STATUS, value)

class TuyaGasDetector0601(CustomDevice):
    """TS0601 _TZE200_iuk8kupi quirk."""

    def __init__(self, *args, **kwargs):
        """Init."""
        self.ias_bus = Bus()
        super().__init__(*args, **kwargs)

    signature = {
        MODELS_INFO: [
            ("_TZE200_iuk8kupi", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zigpy.profiles.zha.PROFILE_ID,
                DEVICE_TYPE: zigpy.profiles.zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zigpy.profiles.zha.PROFILE_ID,
                DEVICE_TYPE: zigpy.profiles.zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaGasDetectorZone,
                    TuyaGasDetectorCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
        },
    }

javicalle, 2024-03-02:
It seems to me that this would be an old quirk, but you can try to replace the import statement:
```
from zhaquirks.tuya.mcu import TuyaManufCluster, TuyaManufClusterAttributes
```

gijsh070, 2024-03-04:
Thanks for your repply, i tried replacing but still i do not see the sensors or quirk showing up. Probably I need to buy another device that is completely supported in ZHA or I have to move to zigbee2mqtt

gijsh070, 2024-03-04:
Fixed it by moving to mqtt, got way more options for other products as well.

closed 2024-03-04 by gijsh070

zolakt, 2025-07-29:
I got this device working in zha with the following v2 quirk:

```
from zigpy.types import t
from zigpy.quirks.v2.homeassistant import EntityType, CONCENTRATION_PARTS_PER_MILLION
from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass, SensorStateClass
from zigpy.quirks.v2.homeassistant.binary_sensor import BinarySensorDeviceClass
from zigpy.zcl import foundation
from zhaquirks.tuya import TUYA_CLUSTER_ID
from zhaquirks.tuya.builder import TuyaQuirkBuilder

(
    TuyaQuirkBuilder(
        manufacturer="_TZE200_iuk8kupi",
        model="TS0601",
    )
    # CH4 Alarm
    .tuya_dp_attribute(
        dp_id=1,
        attribute_name="ch4_alarm",
        type=t.Bool,
        access=foundation.ZCLAttributeAccess.Read | foundation.ZCLAttributeAccess.Report,
        converter=lambda x: 1 if x == 0 else 0,
    )
    .binary_sensor(
        attribute_name="ch4_alarm",
        cluster_id=TUYA_CLUSTER_ID,
        endpoint_id=1,
        entity_type=EntityType.STANDARD,
        device_class=BinarySensorDeviceClass.GAS,
        initially_disabled=False,
        attribute_initialized_from_cache=True,
        translation_key=None,
        fallback_name="Methane Alarm",
    )
    # CH4 Concentration sensor
    .tuya_sensor(
        dp_id=2,
        attribute_name="ch4_concentration",
        type=t.int16s,
        device_class=SensorDeviceClass.VOLATILE_ORGANIC_COMPOUNDS_PARTS,
        state_class=SensorStateClass.MEASUREMENT,
        unit=CONCENTRATION_PARTS_PER_MILLION,
        fallback_name="Methane"
    )
    # CO Alarm
    .tuya_dp_attribute(
        dp_id=18,
        attribute_name="co_alarm",
        type=t.Bool,
        access=foundation.ZCLAttributeAccess.Read | foundation.ZCLAttributeAccess.Report,
        converter=lambda x: 1 if x == 0 else 0,
    )
    .binary_sensor(
        attribute_name="co_alarm",
        cluster_id=TUYA_CLUSTER_ID,
        endpoint_id=1,
        entity_type=EntityType.STANDARD,
        device_class=BinarySensorDeviceClass.CO,
        initially_disabled=False,
        attribute_initialized_from_cache=True,
        translation_key=None,
        fallback_name="Carbon Monoxide Alarm",
    )
    # CO Concentration sensor
    .tuya_sensor(
        dp_id=19,
        attribute_name="co_concentration",
        type=t.int16s,
        device_class=SensorDeviceClass.CO,
        state_class=SensorStateClass.MEASUREMENT,
        unit=CONCENTRATION_PARTS_PER_MILLION,
        fallback_name="Carbon Monoxide",
        converter=lambda x: x/100,
    )
    .skip_configuration()
    .add_to_registry()
)
```

However, I have huge issues with this device. It floods the network with reports, causing errors, delays of other devices etc. Is that somehow solved in z2m?