zigpy/zha-device-handlers #3016: [BUG] IKEA Buttons having Battery and Pairing Issues On QuirkV2

RSZEE, 2024-03-02:
### Bug description

Whenever my IKEA devices disconnect for no reason (even with a new battery), when I repair them they take 7 tries or evenmore and the battery percentage never shows up.  I've noticed that this has happened after my devices changed their quirks to V2

### Steps to reproduce

ZHA -> IKEA E1812 Button -> Battery
OR
ZHA -> Add Devices

### Expected behavior

To pair properly and to show the battery percentage.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
16
17
18
19
20
21
22
23
24
25
26
27
28
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0820",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0009",
        "0x0020",
        "0x1000"
      ],
[... 455 lines cut ...]
            "unsupported_attributes": {}
          },
          "0x1000": {
            "endpoint_attribute": "lightlink",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

MattWestb, 2024-03-02:
All IKEA controllers is working good **if** having one good router as parent.
So the interesting is your coordinator and its firmware version and all your router in the network.

RSZEE, 2024-03-02:
I have a Sonoff Zigbee 3.0 P as my dongle. this is only happening with ikea buttons

MattWestb, 2024-03-02:
Firmware version and other routers in the network (The TI chip cant handle SEDs if not running current production firmware) ?

RSZEE, 2024-03-17:
Firmware: Z-Stack 20210708
- Sonoff DOngle Firmware

closed 2024-09-20 by github-actions[bot]