zigpy/zha-device-handlers #3017: [Device Support Request] Philips Hue Secure Floodlight Camera EU (929003608801)

JohnConnett, 2024-03-02:
### Problem description

This is a slightly unusual device. It is essentially two devices: a Zigbee Colour Floodlight and a Secure Camera sharing a common base and power supply.

Although the packaging describes it as a [Zigbee Certified Product](https://zigbeealliance.org/wp-content/uploads/2021/04/07-4842-13-Zigbee-certification-policy.pdf) the camera part is unable to pair with ZHA. Nor does it support pairing using Zigbee Direct using Bluetooth LE (BLE).

It can be commissioned into a Zigbee network with a [Philips Hue Bridge](https://www.philips-hue.com/en-gb/p/hue-bridge/8719514342583) as coordinator using the proprietary [Philips Hue App](https://www.philips-hue.com/en-gb/explore-hue/apps/bridge) using WiFi and BLE. The BLE advertisements emitted during setup do not conform to the _BLE Connectable Advertisement example_ in Appendix A of **Zigbee Direct Specification, Version 1.0**. In particular, instead of the 16-bit UUID of the Zigbee Direct Commissioning Service (0xFFF7) it has the 16-bit UUID of Signify Netherlands B.V. (formerly Philips Lighting B.V.) (0xFE0F).

It may not be feasible to support this device with ZHA.

I have set up a GitHub repository [Hue-Secure-Camera](https://github.com/JohnConnett/Hue-Secure-Camera) to collate information about this device.

### Solution description

I would like to be able to pair this **Zigbee Certified Product** with ZHA (or other Zigbee networks of my choice).

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

MattWestb, 2024-03-02:
Is it this certificated product ? 
https://csa-iot.org/csa_product/hue-secure-camera-0124/

And pleas post the diagnostic and / or Device signature

JohnConnett, 2024-03-02:
> Is it this certificated product ? https://csa-iot.org/csa_product/hue-secure-camera-0124/

That was the only Hue Secure Camera entry I could find. My guess is that the Wired, Battery and Floodlight cameras are all covered by the same certification.

> And please post the diagnostic and / or Device signature

Would be happy to if I knew how to obtain them from a device that I can't connect to ZHA.

I have attached the output of `https://<bridge ip address>/clip/v2/resource/device` which might help. The system consists of a Hue Bridge v2.1; the Floodlight Camera and a [Senic Friends of Hue Outdoor Switch](https://www.senic.com/products/friends-of-hue-outdoor) (a convenient way to turn off the floodlight while I'm testing it indoors!). There are some details in the file that may be of use.

[response.json](https://github.com/zigpy/zha-device-handlers/files/14470083/response.json)

Hedda, 2024-08-16:
FYI, noticed that @JohnConnett got it working with Zigbee2MQTT and there is lots more information in the discussion for it there:

* https://github.com/Koenkk/zigbee2mqtt/issues/21650

He also posted a feature request for ZHA in Home Assistant community forum asking for ZHA to support Hue QR install codes:

* https://community.home-assistant.io/t/zha-developer-tools-zha-permit-should-accept-hue-qr-install-codes/705666

@JohnConnett I believe that the request for Hue QR install code for ZHA might instead belong as issue for zigpy or zha libraries:

* https://github.com/zigpy/zigpy/issues

* https://github.com/zigpy/zha/issues

JohnConnett, 2024-08-16:
> FYI, noticed that @JohnConnett got it working with Zigbee2MQTT and there is lots more information in the discussion for it there:
> 
>     * [[New device support]: Philips Hue Secure Floodlight Camera EU (929003608801) Koenkk/zigbee2mqtt#21650](https://github.com/Koenkk/zigbee2mqtt/issues/21650)
Unfortunately, I haven't _**got it working with Zigbee2MQTT**_.

Although it is a _Zigbee Certified Product_ (even printed on the packaging!). That doesn't necessarily mean that you can connect it to a Zigbee network other than one hosted by a Philips Hue Bridge. I will continue to investigate if it is possible to connect it to other Zigbee networks as time permits.

Even if a _Zigbee Certified Product_ can be connected to your choice of Zigbee network there may still be problems if it uses undocumented, manufacturer specific attributes and commands.

(Off topic) There may also be issues with accessing the Wi-Fi features, particularly access to encrypted video and audio streams.

closed 2025-08-14 by github-actions[bot]