zigpy/zha-device-handlers #3018: [BUG] IKEA Vallhorn Update failed

theincogtion, 2024-03-02:
### Bug description

Firmware update is displayed in Home Assistant.
Trying to install it fails with this error Message:
[image]

### Steps to reproduce

Go to settings and try to update vallhorn firmware
[image]

### Expected behavior

Update works

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
2024-03-02 17:34:04.577 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e87000700aa00
2024-03-02 17:34:04.579 DEBUG (MainThread) [zigpy_deconz.api] Received command CommandId.device_state_changed{'status': <Status.SUCCESS: 0>, 'frame_length': 7, 'device_state': DeviceState(network_state=<NetworkState2.CONNECTED: 2>, device_state=<DeviceStateFlags.APSDE_DATA_INDICATION|APSDE_DATA_REQUEST_FREE_SLOTS_AVAILABLE|32: 42>), 'reserved': 0} (seq 135)
2024-03-02 17:34:04.583 DEBUG (MainThread) [zigpy_deconz.api] Sending CommandId.aps_data_indication{'flags': <DataIndicationFlags.Always_Use_NWK_Source_Addr: 1>} (seq=135)
2024-03-02 17:34:04.584 DEBUG (MainThread) [zigpy_deconz.uart] Send: 1787000800010001
2024-03-02 17:34:04.587 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x17870026001f002202000001020e4003040100040800184b0a0000214d1b00afffbd7bbd00b0
2024-03-02 17:34:04.590 DEBUG (MainThread) [zigpy_deconz.api] Received command CommandId.aps_data_indication{'status': <Status.SUCCESS: 0>, 'frame_length': 38, 'payload_length': 31, 'device_state': DeviceState(network_state=<NetworkState2.CONNECTED: 2>, device_state=<DeviceStateFlags.APSDE_DATA_REQUEST_FREE_SLOTS_AVAILABLE: 8>), 'dst_addr': DeconzAddress(address_mode=<AddressMode.NWK: 2>, address=0x0000), 'dst_ep': 1, 'src_addr': DeconzAddress(address_mode=<AddressMode.NWK: 2>, address=0x400E), 'src_ep': 3, 'profile_id': 260, 'cluster_id': 1024, 'asdu': b'\x18K\n\x00\x00!M\x1b', 'reserved1': 0, 'reserved2': 175, 'lqi': 255, 'reserved3': 189, 'reserved4': 123, 'reserved5': 189, 'reserved6': 0, 'rssi': -80} (seq 135)
2024-03-02 17:34:04.591 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2024, 3, 2, 16, 34, 4, 591288, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x400E), src_ep=3, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=1024, data=Serialized[b'\x18K\n\x00\x00!M\x1b'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-80)
2024-03-02 17:34:04.592 DEBUG (MainThread) [zigpy.zcl] [0x400E:3:0x0400] Received ZCL frame: b'\x18K\n\x00\x00!M\x1b'
2024-03-02 17:34:04.593 DEBUG (MainThread) [zigpy.zcl] [0x400E:3:0x0400] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl<0x18>(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Server_to_Client: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=75, command_id=10, *direction=<Direction.Server_to_Client: 1>)
2024-03-02 17:34:04.594 DEBUG (MainThread) [zigpy.zcl] [0x400E:3:0x0400] Decoded ZCL frame: IlluminanceMeasurement:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint16_t, value=6989))])
2024-03-02 17:34:04.594 DEBUG (MainThread) [zigpy.zcl] [0x400E:3:0x0400] Received command 0x0A (TSN 75): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint16_t, value=6989))])
2024-03-02 17:34:04.595 DEBUG (MainThread) [zigpy.zcl] [0x400E:3:0x0400] Attribute report received: measured_value=6989
2024-03-02 17:34:04.595 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0x400E:3:0x0400]: cluster_handler[illuminance] attribute_updated - cluster[Illuminance Measurement] attr[measured_value] value[6989]
2024-03-02 17:34:04.597 DEBUG (Thread-5) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f73a55030>, 'UPDATE devices_v12\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_update_delta', {'ts': 1709397244.591288, 'ieee': 8c:6f:b9:ff:fe:1c:0c:51, 'min_update_delta': 30.0})
2024-03-02 17:34:04.598 DEBUG (Thread-5) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f73a55030>, 'UPDATE devices_v12\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_update_delta', {'ts': 1709397244.591288, 'ieee': 8c:6f:b9:ff:fe:1c:0c:51, 'min_update_delta': 30.0}) completed
2024-03-02 17:34:04.600 DEBUG (Thread-5) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f73a55030>)
2024-03-02 17:34:04.600 DEBUG (Thread-5) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f73a55030>) completed
2024-03-02 17:34:04.601 DEBUG (Thread-5) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f73a55030>, '\n            INSERT INTO attributes_cache_v12\n            VALUES (:ieee, :endpoint_id, :cluster_id, :attrid, :value, :timestamp)\n                ON CONFLICT (ieee, endpoint_id, cluster, attrid) DO UPDATE\n                SET value=excluded.value, last_updated=excluded.last_updated\n                WHERE\n                    value != excluded.value\n                    OR :timestamp - last_updated > :min_update_delta\n            ', {'ieee': 8c:6f:b9:ff:fe:1c:0c:51, 'endpoint_id': 3, 'cluster_id': 1024, 'attrid': 0, 'value': 6989, 'timestamp': 1709397244.595276, 'min_update_delta': 30.0})
2024-03-02 17:34:04.602 DEBUG (Thread-5) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f73a55030>, '\n            INSERT INTO attributes_cache_v12\n            VALUES (:ieee, :endpoint_id, :cluster_id, :attrid, :value, :timestamp)\n                ON CONFLICT (ieee, endpoint_id, cluster, attrid) DO UPDATE\n                SET value=excluded.value, last_updated=excluded.last_updated\n                WHERE\n                    value != excluded.value\n                    OR :timestamp - last_updated > :min_update_delta\n            ', {'ieee': 8c:6f:b9:ff:fe:1c:0c:51, 'endpoint_id': 3, 'cluster_id': 1024, 'attrid': 0, 'value': 6989, 'timestamp': 1709397244.595276, 'min_update_delta': 30.0}) completed
2024-03-02 17:34:04.603 DEBUG (Thread-5) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f73a55030>)
2024-03-02 17:34:04.603 DEBUG (Thread-5) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f73a55030>) completed
2024-03-02 17:34:07.610 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e88000700aa00
[... 105 lines cut ...]
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 892, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 962, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 161, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 465, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/zha/update.py", line 177, in async_install
    raise HomeAssistantError(f"Update was not successful - result: {results}")
homeassistant.exceptions.HomeAssistantError: Update was not successful - result: <Status.FAILURE: 1>
2024-03-02 17:34:24.190 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x10AA](TRADFRI remote control): Device seen - marking the device available and resetting counter
2024-03-02 17:34:24.191 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x10AA](TRADFRI remote control): Update device availability -  device available: True - new availability: True - changed: False
```

</details>

### Additional information

Also found here:
https://community.home-assistant.io/t/zigbee-firmware-update-failed/687459/18

MattWestb, 2024-03-02:
Duplicate of https://github.com/zigpy/zigpy/issues/1347 but good that im not alone !!
I have one that was being updated but its in one test system with RCP coordinator and the tricky one is in my production system.

theincogtion, 2024-03-02:
I am following this now:
[image]
And I am one step ahead now:
[image]
I removed the batteries, places it near my conbee and reinserted them.

puddly, 2024-03-02:
What brand of router device is it joined through? It looks like a request is being sent to the device but it doesn't respond within 10s.

theincogtion, 2024-03-02:
> What brand of router device is it joined through? It looks like a request is being sent to the device but it doesn't respond within 10s.
Conbee 2 (usb with mini cable on raspi 4)
[image]

Live Ticker:
[image]

MattWestb, 2024-03-02:
For me the working one:
Device info
LEPTITER Recessed spot light
by IKEA of Sweden
Connected via [Billy RCP 7.4.0.0 RK3318
Firmware: 0x23087631 (Gen 1/2)

And the not working is on:
Device info
STOFTMOLN ceiling/wall lamp WW24
by IKEA of Sweden
Firmware: 0x00010006 (Gen 3).

devosthomas, 2024-03-28:
This also worked for me: remove batteries, put them back in near the Conbee II USB dongle and launched the update process. The update takes time so be patient.

superami-code, 2024-07-12:
I have two of these Vallhorn motion sensors. The one is still on the older firmware and it works fine. The other got the firmware update and now it will connect, but I can't bind to it, so there are no updates after the initial connection.

Is there a way to rerun the firmware update or downgrade it to the old firmware?

closed 2025-01-15 by github-actions[bot]