zigpy/zha-device-handlers #3027: [Device Support Request]  Gas leak detector TS0601 by _TZE200_nus5kk3n

onmobs, 2024-03-07:
### Problem description

Hello. I recently bought gas leak detector, sadly after pairing it shows no sensors. 
 

### Solution description

I guess quirk will do)

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_nus5kk3n",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

PawelGribov, 2024-04-15:
I have the same sensor, it works with this quirk 
[ts0601_gas.txt](https://github.com/zigpy/zha-device-handlers/files/14982935/ts0601_gas.txt)

If I press test button then gas binary sensor entity doesn't react, but it reacts on real gas: the gas detector flashes red and beeps.

If the gas concentration is weak, the detector may simply flash red, but not emit a sound signal, in that case the binary sensor also doesn't respond.

onmobs, 2024-04-16:
Thanks for info, but I already moved to z2m)

closed 2024-10-20 by github-actions[bot]