zigpy/zha-device-handlers #3030: [IMPROVEMENT] convert Legrand 064882 cable outlet to climate entity instead of switch

luke7101, 2024-03-10:
### Problem description

Following https://github.com/zigpy/zha-device-handlers/pull/2807 I added the device, but it's discovered as a switch entity instead of a climate entity, despite its purpose.
I think as it's designed to drive a electric heater with (or without) pilot wire it should be detected as a climate entity.

### Solution description

The device can be in a standard (on/off) mode or a pilot wire mode (6 orders).
Standard is set by default or by writing ``10`` in ``LegrandCluster 0XFC01`` ``device_mode 0X0000``
Pilot wire mode is set by by writing ``20`` in ``LegrandCluster 0XFC01`` ``device_mode 0X0000``

The climate entity could have ``hvac_modes`` set to ``off``, ``on`` or ``pilot wire``.
Off and on should control the on/off switch in the standard mode
Pilot wire should set the device to the pilot wire mode.
Example:
[image]

Then ``preset_modes`` would be selectable between the 6 available pilot wire orders in ``LegrandCableOutletCluster 0XFC40`` ``device_mode 0X0000`` by sending the following commands: ``COMFORT = 0x00`` ``COMFORT_MINUS_1 = 0x01`` ``COMFORT_MINUS_2 = 0x02`` ``ECO = 0x03`` ``FROST_PROTECTION = 0x04`` ``OFF = 0x05``
Example:
[image]

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

bemble, 2024-03-11:
What I did was the first step to getting it working for the winter, there's room for improvement for sure, and this will come in a quite close future.

luke7101, 2024-03-11:
And it works! Thanks very much 😊

Looking forward for it !

bemble, 2024-03-11:
> And it works! Thanks very much 😊
> 
> Looking forward for it !

Take a look at #3031 ;)

luke7101, 2024-03-11:
> Take a look at [#3031](https://github.com/zigpy/zha-device-handlers/pull/3031) ;)

Perfect timing ! 😁

closed 2024-09-17 by github-actions[bot]