zigpy/zha-device-handlers #3040: [Device Support Request] SIN-4-FP-21 - NodOn

Mookunicorn, 2024-03-13:
### Problem description

Missing controls the target temperature of the heater with possible values: comfort, eco, frost_protection, off, comfort_-1, comfort_-2. Currently just an "on, off" switch

### Solution description

Add controls of the heater
(comfort, eco, frost_protection, off, comfort_-1, comfort_-2.)

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4747, maximum_buffer_size=82, maximum_incoming_transfer_size=500, server_mask=11264, maximum_outgoing_transfer_size=500, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x1000",
        "0xfc00"
      ],
      "output_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0066",
      "input_clusters": [
        "0x0021"
      ],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "NodOn",
  "model": "SIN-4-FP-21",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.3.0",
[... 705 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
Sorry I didn't manage to make it, obviously I'm not focused enough to understand how to make it. But I'm trying and if I find how to do it, I promise I'll post it here.
```

</details>

Skunnyk, 2024-03-18:
Related: https://github.com/zigpy/zha-device-handlers/issues/2777

Kriss1670, 2024-03-21:
Waiting also for ZHA support...

oxillo, 2024-05-13:
I've just made some progress on it using quirks v2. It is my first quirk and I'm new to HA so I believe there is room for improvement.

Here is the quirk
```python
"""NodOn on/off switch two channels."""
from typing import Final
from zigpy.quirks import CustomCluster
from zigpy.quirks.v2 import add_to_registry_v2, EntityType, EntityPlatform
from zigpy.zcl.foundation import (
    BaseAttributeDefs,
    BaseCommandDefs,
    ZCLAttributeDef,
    ZCLCommandDef
)

from zigpy.zcl import ClusterType

import zigpy.types as t

class NodonPilotWireMode(t.enum16):
    OFF = 0x00
    COMFORT = 0x01
    ECO = 0x02
    FROST_PROTECTION = 0x03
    COMFORT_MINUS_1 = 0x04
    COMFORT_MINUS_2 = 0x05

NODON_PILOT_WIRE_CLUSTER_ID = 0xfc00  #64512

class NodonPilotWireCluster(CustomCluster):
    """NodOn pilot wire manufacturer-specific cluster."""

    cluster_id: Final = NODON_PILOT_WIRE_CLUSTER_ID
    name: Final = "Pilot Wire"
    ep_attribute: Final = "pilot_wire_cluster"
    
    
    class AttributeDefs(BaseAttributeDefs):
        pilot_wire_mode: Final = ZCLAttributeDef(id=0x0000, type=NodonPilotWireMode, access="r")

    class ClientCommandDefs(BaseCommandDefs):
        setMode: Final = ZCLCommandDef(
            id=0x0000,
            schema={
                "mode": NodonPilotWireMode
            },
            direction=True,
            is_manufacturer_specific=True,
        )

(
    add_to_registry_v2("NodOn", "SIN-4-FP-21")
    .replaces(NodonPilotWireCluster)
    .enum("pilot_wire_mode",NodonPilotWireMode,NODON_PILOT_WIRE_CLUSTER_ID,
          ClusterType.Server,1,EntityType.STANDARD,EntityPlatform.SELECT,False,True,"pilot_wire_mode")
)
```

And the result in entity view
[image]

oxillo, 2024-05-13:
And with an addition `also_applies_to()` that should solve #2777 

````python
(
    add_to_registry_v2("NodOn", "SIN-4-FP-21")
    .also_applies_to("ADEO", "SIN-4-FP-21_EQU")
    .replaces(NodonPilotWireCluster)
    .enum("pilot_wire_mode",NodonPilotWireMode,NODON_PILOT_WIRE_CLUSTER_ID,
          ClusterType.Server,1,EntityType.STANDARD,EntityPlatform.SELECT,False,True,"pilot_wire_mode")
)
```

oxillo, 2024-05-14:
Hello, 
I'm using the quirk on the latest version of HA (2024.5.3). What version do you use ? 
Quirks v2 are new and support is evolving at every version.

Mookunicorn, 2024-05-14:
I'm on 2024.5.3 too 🤔
[image]

Mookunicorn, 2024-05-14:
I wonder if the problem could be with the ZigBee key.

I use SkyConnect without multiprotocol

oxillo, 2024-05-18:
This is not your installation; it is my solution. 
The enum entity tries to write the `pilot_wire_mode` attribute which is not supported. You must use the command to change the mode.

oxillo, 2024-05-19:
Some update to catch attribute write and replace by commands 

```python
"""NodOn SIN-4-FP-21 pilot wire module."""
from typing import Any, Final
from zigpy.quirks import CustomCluster
from zigpy.quirks.v2 import add_to_registry_v2, EntityType, EntityPlatform
from zigpy.zcl.foundation import (
    BaseAttributeDefs,
    BaseCommandDefs,
    ZCLAttributeDef,
    ZCLCommandDef,
    WriteAttributesStatusRecord,
    Status
)

from zigpy.zcl import ClusterType

import zigpy.types as t

import logging
_LOGGER = logging.getLogger(__name__)

class NodonPilotWireMode(t.enum16):
    OFF = 0x00
    COMFORT = 0x01
    ECO = 0x02
    FROST_PROTECTION = 0x03
    COMFORT_MINUS_1 = 0x04
    COMFORT_MINUS_2 = 0x05

NODON_PILOT_WIRE_CLUSTER_ID = 0xfc00  #64512

class NodonPilotWireCluster(CustomCluster):
    """NodOn pilot wire manufacturer-specific cluster."""

    cluster_id: Final = NODON_PILOT_WIRE_CLUSTER_ID
    name: Final = "Pilot Wire"
    ep_attribute: Final = "pilot_wire_cluster"

    class AttributeDefs(BaseAttributeDefs):
        pilot_wire_mode: Final = ZCLAttributeDef(id=0x0000, type=NodonPilotWireMode, access="r")

    class ClientCommandDefs(BaseCommandDefs):
        setMode: Final = ZCLCommandDef(
            id=0x0000,
            schema={"mode": NodonPilotWireMode},
            direction=True,
            is_manufacturer_specific=True,
        )
    
    async def write_attributes(
        self, attributes: dict[str | int, Any], manufacturer: int | None = None
    ) -> list:
        """Override writes to the pilot_wire_mode attribute."""
        if "pilot_wire_mode" in attributes:
            await self.client_command(0,attributes.get("pilot_wire_mode"))
            return [[WriteAttributesStatusRecord(Status.SUCCESS)]]
        return await super().write_attributes(attributes, manufacturer=manufacturer)

(
    add_to_registry_v2("NodOn", "SIN-4-FP-21")
    .also_applies_to("ADEO", "SIN-4-FP-21_EQU")
    .replaces(NodonPilotWireCluster)
    .enum("pilot_wire_mode",NodonPilotWireMode,NODON_PILOT_WIRE_CLUSTER_ID,
        ClusterType.Server,1,EntityType.STANDARD,EntityPlatform.SELECT,False,True)
)
```

oxillo, 2024-05-28:
Happy to see it works for you !
You can add the quirk to your initial request.
I would like to work on the HMI but I have other projects in parallel that requires my attention. I'll look into it in some weeks.

rungeard, 2024-07-20:
> any idea how to avoid this problem ?

@RenZan it seems to be related to https://github.com/zigpy/zigpy/issues/1410

anoril1979, 2024-09-18:
Hi!
And update from @garc33
https://github.com/zigpy/zha-device-handlers/issues/2777#issuecomment-2358979497

oxillo, 2024-09-22:
Thanks @ikruglov for initiating a PR. 
I did not initiated it by myself as I'm quite new to HA and thought that my additions were not mature enough.
Indeed, I would like that the pilot wire mode entity to have a name as "select.radiateur_sam _pilotwiremode" and not as "select.radiateur_sam_none" as of today. i would appreciate any help of this.
I would also like to have a default card for pilot wire.

For QuirkBuilder, it is the new way to define quirks (v2). The PR #3364 uses the old style; so it should not be needed.

ikruglov, 2024-09-22:
Thanks for the clarification @oxillo. Let's join forces to get it well-prepared and merged.

I have a question. Why did you need special handing of `write_attributes()`?  In my case, I just defined an attribute and a command. As a result, both setting and getting pilot_wire modes work without any extra handling.

Also, I've tried using the v2 interface with QuirkBuilder, but tests have started failing. I haven't found a way to keep them working so far. So, I'll probably keep the code in its current state unless the code owner objects.

ikruglov, 2024-09-22:
@oxillo I've re-created my PR. It now uses `QuirkBuilder()`. I decided to use it because it has a much cleaner interface and very concise code. I still lack tests, but I've asked for maintainer help with an example.

The code works in my case. I can get and set pilot wire mode in both the HomeAsistant interface and directly manipulating cluster via the 'Manage Zigbee device.' It required no extra code for me or specific handling.

I also understood what you mean by "radiateur_sam_none." Yeah, I also didn't find a way. But it's a simple problem. The entity can be renamed in the HA interface, so I'll leave it for now.

jyrivallan, 2024-09-22:
@ikruglov may I ask you to take into account https://github.com/zigpy/zha-device-handlers/issues/2777 at the same time? `SIN-4-FP-21_EQU` and `SIN-4-FP-21` are exactly the same device.
I'm no expert, but it seems it could be as simple as a `.also_applies_to("NodOn","SIN-4-FP-21")` in the QuirkBuilder

ikruglov, 2024-09-22:
@jyrivallan sure thing. I've added it to the PR. But I have no way of testing it. So, you will need to do it.

oxillo, 2024-09-22:
@ikruglov : About the write_attributes() , it is a trick to be able to have a combobox to control the pilot wire mode in the ZHA entity page (see picture in #3040). 
We could also have just the mode value displayed and 6 buttons to set each individual mode (one for OFF, one for COMFORT, one for ECO,...)

garc33, 2024-09-22:
> @jyrivallan sure thing. I've added it to the PR. But I have no way of testing it. So, you will need to do it.

I comment in the PR it won't work. The adeo device needs to override the manufacturer code.

ikruglov, 2024-09-22:
> @ikruglov : About the write_attributes() , it is a trick to be able to have a combobox to control the pilot wire mode in the ZHA entity page (see picture in #3040). We could also have just the mode [...]

@oxillo thanks for the clarification. With my PR I'm getting the sa

[image]
me result I think.  And no extra handling.

ikruglov, 2024-09-22:
> > @jyrivallan sure thing. I've added it to the PR. But I have no way of testing it. So, you will need to do it.
> 
> I comment in the PR it won't work. The adeo device needs to override the manufacturer code.

@garc33 I don't see your comment. Where did you leave it?

garc33, 2024-09-23:
> > > @jyrivallan sure thing. I've added it to the PR. But I have no way of testing it. So, you will need to do it.
> 
> > 
> 
> > I comment in the PR it won't work. The adeo device needs to override the manufacturer code.
> 
> 
> 
> @garc33 I don't see your comment. Where did you leave it?

https://github.com/zigpy/zha-device-handlers/pull/3364#discussion_r1770588689

ikruglov, 2024-09-23:
> > > > @jyrivallan sure thing. I've added it to the PR. But I have no way of testing it. So, you will need to do it.
> > 
> > 
> > > 
> > 
> > 
> > > I comment in the PR it won't work. The adeo device needs to override the manufacturer code.
> > 
> > 
> > @garc33 I don't see your comment. Where did you leave it?
> 
> [#3364 (comment)](https://github.com/zigpy/zha-device-handlers/pull/3364#discussion_r1770588689)

For some reason, I don't see this comment.

If a one-line change doesn't work, I will revert it since I cannot test changes myself. Someone with the ADEO device is welcome to follow up on my PR.

anoril1979, 2024-09-29:
Hello!
I tried this quirk.
The device attribute names changed, so I guess the quirk is applied, but I don't have any selector for the pilot wire mode.

anoril1979, 2024-09-29:
Hello!
I'm struggling with Home Assistant for this one!
I added the quirk and restarted HA. Then I unpaired my devices and refresh. For a moment, they were there, displaying the pilotwire mode and I thought it was ok. But changing the mode did nothing on the device.
I refreshed and restarted and went into "multipledeviceexception" on ZHA side. I realized my devices were all named "SIN-4..." I removed them and restarted. Impossible to get ride of them.
Well, in the end, I refreshed, restarted and now I have my 3 devices back.
But they are considered as "zigpy.device.Device" and the quirk is not dealing with them.
I'm a bit lost.
Note: the device work fine: I can commute and have power consumption feedback. But no wire mode.
[image]
[image]
[image]

anoril1979, 2024-09-29:
I removed all my Adeo devices. I restarted HA by rebooting the rapsberry.
Hmmm, my devices are in pairing mode (by themself, weird) and "search new devices" does not seems to find them anymore.

Mookunicorn, 2024-11-07:
Since 2024.11.0 HA update quirk doesn't work.

From what I understand from the logs, the attribute "'fallback_name'" is missing in ".enum". I therefore tried adding it as follows without success:
```python
.enum("pilot_wire_mode", NodonPilotWireMode, NODON_PILOT_WIRE_CLUSTER_ID,
    ClusterType.Server, 1, EntityType.STANDARD, EntityPlatform.SELECT, False, True, fallback_name="mode")
```

<details><summary>Error Log</summary>
Enregistreur: zhaquirks
Source: /usr/local/lib/python3.12/site-packages/zhaquirks/__init__.py:481
S'est produit pour la première fois: 10:49:36 (1 occurrences)
Dernier enregistrement: 10:49:36

Unexpected exception importing custom quirk 'NodOnPiloteWire'
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/zhaquirks/__init__.py", line 479, in setup
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_zha_quirks/NodOnPiloteWire.py", line 78, in <module>
    .enum(
     ^^^^^
  File "/usr/local/lib/python3.12/site-packages/zigpy/quirks/v2/__init__.py", line 666, in enum
    ZCLEnumMetadata(
  File "<attrs generated init zigpy.quirks.v2.ZCLEnumMetadata>", line 16, in __init__
    __attr_validator_fallback_name(self, __attr_fallback_name, self.fallback_name)
  File "/usr/local/lib/python3.12/site-packages/attr/validators.py", line 101, in __call__
    raise TypeError(
TypeError: ("'fallback_name' must be <class 'str'> (got None that is a <class 'NoneType'>).", Attribute(name='fallback_name', default=NOTHING, validator=<instance_of validator for type <class 'str'>>, repr=True, eq=True, eq_key=None, order=True, order_key=None, hash=None, init=True, metadata=mappingproxy({}), type='str', converter=None, kw_only=True, inherited=True, on_setattr=None, alias='fallback_name'), <class 'str'>, None)
</details>
<details><summary>Custom quirk</summary>

```python
"""Nodon SIN-4-FP-21 | Adeo SIN-4-FP-21_EQU"""
from typing import Any, Final
from zigpy.quirks import CustomCluster
from zigpy.quirks.v2 import QuirkBuilder, EntityType, EntityPlatform
from zigpy.zcl.foundation import (
    BaseAttributeDefs,
    BaseCommandDefs,
    ZCLAttributeDef,
    ZCLCommandDef,
    WriteAttributesStatusRecord,
    Status
)

from zigpy.zcl import ClusterType

import zigpy.types as t

import logging
_LOGGER = logging.getLogger(__name__)

class NodonPilotWireMode(t.enum16):
    OFF = 0x00
    COMFORT = 0x01
    ECO = 0x02
    FROST_PROTECTION = 0x03
    COMFORT_MINUS_1 = 0x04
    COMFORT_MINUS_2 = 0x05

NODON_PILOT_WIRE_CLUSTER_ID = 0xfc00  #64512

class NodonPilotWireCluster(CustomCluster):
    """NodOn pilot wire manufacturer-specific cluster."""

    cluster_id: Final = NODON_PILOT_WIRE_CLUSTER_ID
    name: Final = "Pilot Wire"
    ep_attribute: Final = "pilot_wire_cluster"
    manufacturer_id_override: Final = 4747
    
    class AttributeDefs(BaseAttributeDefs):
        pilot_wire_mode: Final = ZCLAttributeDef(
            id=0x0000, 
            type=NodonPilotWireMode, 
            access="r"
        )

    class ClientCommandDefs(BaseCommandDefs):
        setMode: Final = ZCLCommandDef(
            id=0x0000,
            schema={"mode": NodonPilotWireMode},
            direction=True,
            is_manufacturer_specific=True
        )
    
    async def write_attributes(
        self, attributes: dict[str | int, Any], manufacturer: int | None = None
    ) -> list:
        """Override writes to the pilot_wire_mode attribute."""
        if "pilot_wire_mode" in attributes:
            await self.client_command(0, attributes.get("pilot_wire_mode"), manufacturer=4747)
            await self.read_attributes(attributes)
            return [[WriteAttributesStatusRecord(Status.SUCCESS)]]
        return await super().write_attributes(attributes, manufacturer=4747)

    async def read_attributes(
        self,
        attributes: list[int | str],
        allow_cache: bool = False,
        only_cache: bool = False,
        manufacturer: int | t.uint16_t | None = None,
    ) -> Any:
        return await super().read_attributes(attributes, allow_cache=False, only_cache=False, manufacturer=4747)

(

    QuirkBuilder("ADEO", "SIN-4-FP-21_EQU")
    .also_applies_to("NodOn","SIN-4-FP-21")
    .replaces(NodonPilotWireCluster)
    .enum("pilot_wire_mode",NodonPilotWireMode,NODON_PILOT_WIRE_CLUSTER_ID,
        ClusterType.Server,1,EntityType.STANDARD,EntityPlatform.SELECT,False,True,)

    .add_to_registry()
)
```
</details>

ikruglov, 2024-11-07:
@Mookunicorn check https://github.com/zigpy/zha-device-handlers/pull/3364. I've pushed a fix for 2024.11.0. It works in my setup

ikruglov, 2024-11-27:
FYI: https://github.com/zigpy/zha-device-handlers/pull/3364 got merged. It supports both NodOn and Adeo. The code should become part of the 2024.12 release. Unfortunately, the change comes with only ZHA attributes (one which changes via "Manage Zigbee Device") and not with enum in the HA interface due to concerns from maintainers.

maximelefrancois86, 2024-12-21:
OK so if we want this, for now, we need to remove it from the device registry and add our own quirk.

This works with version 

```python
"""Adeo SIN-4-FP-21_EQU pilot wire module."""
from typing import Any, Final
from zigpy.quirks import CustomCluster, DEVICE_REGISTRY
from zigpy.quirks.v2 import QuirkBuilder, EntityType, EntityPlatform
from zigpy.zcl.foundation import (
    BaseAttributeDefs,
    BaseCommandDefs,
    ZCLAttributeDef,
    ZCLCommandDef,
    WriteAttributesStatusRecord,
    Status
)

from zigpy.zcl import ClusterType

import zigpy.types as t

import logging
_LOGGER = logging.getLogger(__name__)

del DEVICE_REGISTRY._registry_v2[("NodOn","SIN-4-FP-21")]

class NodonPilotWireMode(t.enum16):
    off = 0x00
    confort = 0x01
    eco = 0x02
    hors_gel = 0x03
    confort_moins_1 = 0x04
    confort_moins_2 = 0x05

NODON_PILOT_WIRE_CLUSTER_ID = 0xfc00  #64512

class NodonPilotWireCluster(CustomCluster):
    """NodOn pilot wire manufacturer-specific cluster."""

    cluster_id: Final = NODON_PILOT_WIRE_CLUSTER_ID
    name: Final = "Pilot Wire"
    ep_attribute: Final = "pilot_wire_cluster"
    manufacturer_id_override: Final = 4747
    
    class AttributeDefs(BaseAttributeDefs):
        pilot_wire_mode: Final = ZCLAttributeDef(
            id=0x0000, 
            type=NodonPilotWireMode, 
            access="r"
        )

    class ClientCommandDefs(BaseCommandDefs):
        setMode: Final = ZCLCommandDef(
            id=0x0000,
            schema={"mode": NodonPilotWireMode},
            direction=True,
            is_manufacturer_specific=True
        )
    
    async def write_attributes(
        self, attributes: dict[str | int, Any], manufacturer: int | None = None
    ) -> list:
        """Override writes to the pilot_wire_mode attribute."""
        if "pilot_wire_mode" in attributes:
            await self.client_command(0, attributes.get("pilot_wire_mode"), manufacturer=NodonPilotWireCluster.manufacturer_id_override)
            await self.read_attributes(attributes)
            return [[WriteAttributesStatusRecord(Status.SUCCESS)]]
        return await super().write_attributes(attributes, manufacturer=NodonPilotWireCluster.manufacturer_id_override)

    async def read_attributes(
        self,
        attributes: list[int | str],
        allow_cache: bool = False,
        only_cache: bool = False,
        manufacturer: int | t.uint16_t | None = None,
    ) -> Any:
        return await super().read_attributes(attributes, allow_cache=False, only_cache=False, manufacturer=NodonPilotWireCluster.manufacturer_id_override)

(

    QuirkBuilder("NodOn","SIN-4-FP-21")
    .replaces(NodonPilotWireCluster)
    .enum(
        attribute_name="pilot_wire_mode",
        enum_class=NodonPilotWireMode,
        cluster_id=NODON_PILOT_WIRE_CLUSTER_ID,
        cluster_type=ClusterType.Server,
        endpoint_id=1,
        entity_platform=EntityPlatform.SELECT,
        entity_type=EntityType.STANDARD,
        initially_disabled=False,
        attribute_initialized_from_cache=True,
        reporting_config=None,
        translation_key="mode_fil_pilote",
        fallback_name="mode fil pilote")
    .add_to_registry()
)

```

this works with:

Core 2024.12.5
Supervisor 2024.12.0
Operating System 14.1
User interface 20241127.8