zigpy/zha-device-handlers #3041: [Device Support Request] Tuya _TZE200_cirvgep4

dekar91, 2024-03-15:
### Problem description

The Quirk from https://github.com/zigpy/zha-device-handlers/issues/1944 works partually for me, apart from the battery and firmware fields. I would like to have it fixed.

### Solution description

Battery level should be displayed.
Zigbee firmware should also be exposed, if possible.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]
</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0302",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0402",
        "0x0405",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_cirvgep4",
  "model": "TS0601",
  "class": "ts0601_temperature.TuyaNousE6TempHumiditySensor"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.3.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.12.2",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "6.1.73-haos-raspi",
    "supervisor": "2024.03.0",
    "host_os": "Home Assistant OS 12.1",
    "docker_version": "24.0.7",
    "chassis": "embedded",
[... 566 lines cut ...]
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TemperatureHumidityManufCluster,  # Single bus for temp, humidity, and battery
                    TuyaTemperatureMeasurement,
                    TuyaRelativeHumidity,
                    TuyaPowerConfigurationCluster2AAA,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }
```

</details>

jacekk015, 2024-03-16:
Your quirk already support battery state under DP4. Device just doesn't send it.
Take out battery, wait few seconds, put it inside. Then probably it will send it.
That's battery operated device and it sends as low data as it can.

dekar91, 2024-03-18:
Thanks for the response @jacekk015!
This is weird, because when the device is connected to "Smart Life"   it updates the battery status.
If your have some debugging guide or docs, so I could check, could you please share it with me?

jacekk015, 2024-03-18:
> If your have some debugging guide or docs, so I could check, could you please share it with me?

You cant find in the logs data that haven't been sent.
If device sends something from DP4 you can be more than sure that quirk will forward this to HA.

SmartLife just caches last read. Same as HA do, it will just show you last read from the cache/database.
You could always try to hack SmartLife <-> Device communication, but I'm pretty sure it's not too wise to often ask battery device about battery status, because that drains battery even more.
All battery driven devices are known to be lazy Zigbee devices.

closed 2024-10-01 by github-actions[bot]